Bouni/kicad-jlcpcb-tools #1: Merge feature cpl_rotations_db.csv of JLCKicadTools

bvernoux, 2021-08-09:
It will be great to merge the feature cpl_rotations_db.csv of https://github.com/matthewlai/JLCKicadTools as I have do not found any way to fix the rotation in cpl in the actual plugin

Bouni, 2021-08-10:
I'll see if I can include that as well!
Thanks for the input 😄

Bouni, 2021-08-10:
@bvernoux I've integrated the fix as you proposed, would you mind testing if the result is valid?

bvernoux, 2021-08-10:
Thanks for the fast update
I will compare with https://github.com/matthewlai/JLCKicadTools using my own cpl_rotations_db.csv (I have added few parts not existing)

Bouni, 2021-08-10:
I did a really quick test with a dummy PCB and the outcome seems perfectly fine!

[image]

Bouni, 2021-08-10:
> Thanks for the fast update
> In fact I have checked the code modification [6de05c9](https://github.com/Bouni/kicad-jlcpcb-tools/commit/6de05c9b75860118101940d37170f480a7517fe6) but what I'm doing is I modify this file feature cpl_rotations_db.csv with my own modifications (as there is always new part not in this file) so it will be better to search that file in current directory for example and if it does not exist to download it

I guess you're right. Maybe we should look for a local file, if not present, download it from GitHub!?

What do you think about that?

bvernoux, 2021-08-10:
Yes it will be perfect like that I could check with my different design with expected results

Bouni, 2021-08-11:
@bvernoux Ok, I've implemented the use of the local file. Highly appreciate that you test it!
Even greater would be if you pretend to let one or more of your boards be made by JLC and check what the preview shows.
That way we could check if everything is correct now (screenshots would be awesome 😎 )

bvernoux, 2021-08-11:
I do not know why but with my KiCad 5.99 Nightly (r22289) on Windows 7 kicad-jlcpcb-tools plugin does not start and it is not displayed
Does your plugin is compatible with Python 2.7 or requires Python 3.x ? (as on KiCad it is a bit of a mess and only latest KiCad 5.99 support Python3 IIRC but it does not support anymore Windows7 ...)

bvernoux, 2021-08-12:
I have updated my computer to Windows10 Pro so I can use latest KiCad Nightly + Python 3.9x and I can now test your amazing JLCPCB plugin so stay tuned I'm doing some tests to check the output for different design with modified cpl_rotations_db.csv and I will provide a feedback

Bouni, 2021-08-13:
I concider this issue as solved, what you write about the workflow belongs into its own issue

closed 2021-08-13 by Bouni