Bouni/kicad-jlcpcb-tools #3: LCSC property gets lost when updating PCB from schematic

Bouni, 2021-08-11:
I realized that whenever I update the PCB from the schematic, the LCSC property of all the footprints gets lost.
This means when you select JLC parts you should not update teh PCB from the schematic or you have to do everything again.
Most likely the properties in the PCB do not get updated, but rather deleted and set to whatever comes from the schematic.
Also the Update schematic from PCB Button does not solve this problem as custom properties seem to not make it into the schematic 😞 

I'll file a bug in the KiCAD repo and see if there is a way to persist or sync the properties.

bvernoux, 2021-08-12:
It will be great to use the Schematic BOM and extract fields with LCSC part number (just by searching "LCSC" in Field) as the BOM selection with LCSC part shall be done in Schematic and not in PCBnew like it is done today in the plugin

Bouni, 2021-08-13:
To my knowledge, there is no way to update the schematic from an PCB Editor plugin (and theres no scripting in the schematic editor at all at the moment).
An manipulation the kicad_sch file from the plugin isn't really appealing to me, that could easily corrupt the entire schematic file.

I opened an [issue](https://gitlab.com/kicad/code/kicad/-/issues/8950) in the KiCAD repo to get more info on how to solve this, but from the comments of the KiCAD team it looks like they disencourage the use of custom properties in the PCB editor at all.
They even mentioned taking that ability from the python scripting in a upcoming release.

So I proposed storing the mapping info (Reference - LCSC number) in a seperate file.

We could handle it like that:

- If a LCSC property is set on the schematic, we always use that
- We store that info into the mapping file as well
- Mappings created from the plugin will get stored in the mapping file
- When we create the BOM, we use the info from the mapping file

What do you think about that?

bvernoux, 2021-08-13:
It is a very good idea / alternative as it seems so far there is no other choice

Bouni, 2021-08-18:
From now on (https://github.com/Bouni/kicad-jlcpcb-tools/commit/ca29b2410fc347491e666abb4a0001d714f2b2d8) we store the part assignments in a csv file and no longer in the LCSC property of the footprint.
That way we can avoid loosing the assignments when we update the pcb from the schematic.

On the other hand we can set the LCSC property on a symbol in the schemtaic editor, which will be transfered into the footprint when we upate the pcb from the schematic. When the plugin is started, we write all LCSC properties into the assignment csv and use that info for generating the BOM.

At the moment, we overwrite assignments in the csv, so if you set a part from the plugin which has a LCSC property on the footprint, this will be the value used in the BOM. 

@bvernoux Would you mind testing this new feature?

bvernoux, 2021-08-18:
I have no access to my PC so I will try it as soon as possible
Could you add also properties (in this new CSV to file) to enable/disable BOM/CPL too as it is lost when we re-import from schematic ...?

Bouni, 2021-08-18:
Sure, I wasn't aware that these are overwritten as well!

Bouni, 2021-08-19:
@bvernoux https://github.com/Bouni/kicad-jlcpcb-tools/commit/79379d6618809d8a63ecc3df8645b54624db347b should solve your feature request 😄 
As soon as you have tested the changes and verified them I'll create a new release!

Bouni, 2021-10-14:
I cannot reproduce this behaviour on Windows 🤔 

I've put an ADXL343 in on of my existing PCBs and gave it the `LCSC` property with value `C9667`:
[image]

In the PCB editor I hit update PCB (F8) and placed the part, when I open the plugin I see the LCSC value in the list as expected:
[image]

Bouni, 2021-10-14:
The same behaviour when I do a test with a new project and put just the ADXL part in the schmeatic, apply the LCSC property load the board and open the plugin.

The only way I was able to reproduce this is if I add a trailing space to the LCSC property name.
Can you verify that you have just "LCSC" as the property name and not "LCSC " or something simmilar?