Bouni/kicad-jlcpcb-tools #4: Support gerber export of KiCad Design for 1, 2, 4 or 6 Layers (today it is limited to 2 Layers)

bvernoux, 2021-08-12:
Today actual plugin export Gerber with 2 Layers.
It will be great to also support in addition 1, 4 and 6 layers as so far JLCPCB support only 1, 2, 4 or 6 layers.

Bouni, 2021-08-13:
Indeed I ignored that fact, but when I started I wasn't even sure if I get this to work for double layer PCBs at all.
I guess I will add this feature soon!

Bouni, 2021-08-18:
@bvernoux 
I added multi layer support, if the selection is set to Auto, the plugin detects the layer count automatically, so the board setup is used. But you can still override that with the selection because KiCAD does not let you setup a single layer board as fasr as I can tell.

closed 2021-08-18 by Bouni

bvernoux, 2021-08-18:
Yes I will do some test on different design 2, 4 and 6 layers and I will provide feedback next week

commit e8b8d8da (2021-08-18): allow for different layer count, fix #4