Bouni/kicad-jlcpcb-tools #7: Exception on python action plugin code

criticalboot, 2021-08-17:
getting the below error when trying to use this plugin

[image]

### version info
Application: Pcbnew
Version: 5.1.10-88a1d61d58~90~ubuntu20.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.4.0-80-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Technology: 7.3.0
    Curl: 7.68.0
    Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON

Bouni, 2021-08-18:
This is because you try to use it with a 5.1.10 Version.
I use the nighly builds myself an therefore developed the plugin to work with those versions.

There are quite a few python API changes between version 5 and version 5.99 (nighlty builds) and with a soon to be released vversion 6 I decided to not make it compatible with both versions, sorry 😐

Bouni, 2021-08-18:
One more remark:
I've just installed the [5.1.10 realease](https://kicad-downloads.s3.cern.ch/index.html?prefix=windows/stable/)(2021-05-01T11:58:15.459Z ) as well as the [5.1.10 testing](https://kicad-downloads.s3.cern.ch/index.html?prefix=windows/testing/5.1/)(2021-08-18T03:08:11.015Z ) version and the [patched 5.1.10](https://kicad-downloads.s3.cern.ch/index.html?prefix=windows/testing/patched/)(2021-07-01T16:56:25.176Z) version on windows.

Neither of them comes with `KICAD_SCRIPTING_PYTHON3=ON` 

So if there's no ready to download binary with python 3 support this ist toatally not worth it.

I guess most users will not build their KiCAD installation from source and set that build flag and I'm not willing to make this working with python 2  💀 for obvious reasons 😄

closed 2021-08-18 by Bouni