Bouni/kicad-jlcpcb-tools #9: Error occurred while using.

HaydenHu, 2021-08-20:
Even if kicad is closed, the plug-in cannot be closed. It can be completely closed only by killing the task process.
[image]

Application: KiCad PCB  (64-bit)

Version: (5.99.0-11689-g04c6b79350), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.74.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 17134), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Aug 12 2021 06:25:14
	wxWidgets: 3.1.5 (wchar_t,STL containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.74.0-DEV
	ngspice: 34
	Compiler: Visual C++ 1928 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

Bouni, 2021-08-20:
The problem is that the xls is not downloaded when the UI tries to load data from it.
I need to automatically download it, if not present.

Bouni, 2021-08-22:
Should be fixed with https://github.com/Bouni/kicad-jlcpcb-tools/commit/cec3c0839cccbf171d9788b978b4ccc850d5665a
@HaydenHu I would highly appreciate if you could verify that!