Bouni/kicad-jlcpcb-tools #14: Exception Traceback

radontec, 2021-08-28:
Hi,

I got this error message right after pressing the J@LC button (Win10, latest 5.99 Kicad)

[image]

Bouni, 2021-08-30:
This is strange! looks like there's a footprint in your boad file that has no reference!?
Can you share your design file? (If you do not want it be here in the issues, send it to me as an email attachment to `bouni AT owee DOT de` )

Bouni, 2021-08-30:
@radontec Please test the latest commit, it should fix this issue. After looking closer I guess that there is a footprint that has no number (is that even possible in KiCAD!?) or GetReference() does not return a footprint reference.

Bouni, 2021-08-30:
@radontec 

I've just did a quick test with the files you've sent me and for me it works with the latest commit and I see the same error when using the prior commit.
I run version (5.99.0-11749-g6427197962), release build, on Win 10 as well.

The "problem" is that there are footprints with `REF**` as reference which I had not in mind when writing the sorting function.

But the latest changes (https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/helpers.py#L10-L20) should fix that.

Can you verify that you run the latest version of the plugin? Also make sure to close KiCad completely (and kill all zombie processes if neccessary) before you try again.

radontec, 2021-08-30:
love it! now it works
Bouni thx a lot for fixing the problem

closed 2021-08-30 by radontec