Bouni/kicad-jlcpcb-tools #15: Add JLCPCB Plugin version(+build date) information

bvernoux, 2021-08-30:
Add JLCPCB Plugin version(+build date) information (from github tag) in the title of the Main window "JLCPCB Tools"

Bouni, 2021-08-31:
At the moment I show the short version of the current commit hash:

[image]

This is far from ideal, but showing the tag isn't better because not every commit results in a release and thus not a tag.
So if I show the most recent tag, the user possibly has a much newer version if the main branch is used.

Any ideas what we can do about that?

bvernoux, 2021-08-31:
I think a good things is to display following details GIT_VERSION (including VERSION_COMMIT number+short HASH) + CHECKIN_DATE:
Proposal of Python3 script `git_version_checking_date.py`:
```
from git import *

git=Repo(search_parent_directories=True).git
print("v" + git.describe(tags=True,always=True,dirty=True,long=True) + " " + git.show(['-s', '--pretty=format:%ai']).partition(' ')[0])
```

Such Python3 Script output (on actual https://github.com/Bouni/kicad-jlcpcb-tools) following result: `v0.0.2-4-g1f49599 2021-08-31`
I think it contains all required information to track the exact version used

Bouni, 2021-08-31:
That would require a properly installed git on the target system. 

On my windows machine i hav git installed (included in cmder) but it's not in PATH (I know, what a mess ...) so calling git will result in an exception. 

I tried what you propose, just with a call to subprocess and realized that we generate a new dependency that way (one that can not be easily solved by calling pip as I [already do](https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/library.py#L13-L23) for missing python libs)

Thats why I came up with reading the .git/FETCH_HEAD file to get the current commit hash.

What could we show as a fallback if no git is installed?

bvernoux, 2021-08-31:
I confirm the dependency with GitPython is clearly a bad point especially for user side to avoid adding a new dependency just for Git Tag/Version...

Ideas to avoid GitPython dependency:
1) Create an `Actions Workflow` called also [GitHub Actions](https://docs.github.com/en/actions) to create a file `version_build.txt` including GitPython version/build_date details after each new commit, so on user side you will just read the content of this `version_build.txt` file and display it and it will do not have any link/requirement towards GitPython ...
   - Drawback about that solution:
      - `.git/workflow` will create/update a file for each commit so that will create a new commit and do "pollution" in git commit history for nothing...
      - Maybe we can update that `version_build.txt` as an `artifact` to avoid to do a commit on github and add "pollution" in git commit ...

2) Use your today hint to read `.git` directory/files and just add the git tag version by parsing `.git\refs\tags` finding the latest tag (with the file date which is the most recent), for the build date just use the file modification date of `.git/FETCH_HEAD`
   - What will be missing is the VERSION_COMMIT number as I do not know how to find it from `.git` directory/sub-dir...
   - Drawback about that solution:
      - I'm not sure it is clean do read `.git` directories/files as the format/content can change in future version of git ...

Bouni, 2021-09-01:
@bvernoux I already concidered the workflow method, but a new commit results in a new hash, so the verion hash in version_build.txt is always 1 behind the actual hash, clearly not ideal in my opinion.
I tried that with pre and post commit hooks with the same result as well.

But I like the idea with the artifact. No ieda how to achive this but will defenitely have a look into it!

The idea of reading the FETCH_HEAD as I do at the moment is clearly 💩 as #23 shows 😅

Bouni, 2022-02-09:
I've added a `VERSION` file which always contains `dev` as version. 
On release I write the release tag `2022.02.04` for example into that file and package it into the zip that is used by PCM.
That way via PCM installed versions show the version, installations via cloned git repo show just dev.
Not perfect but good enough for now.

closed 2022-02-09 by Bouni