Bouni/kicad-jlcpcb-tools #16: Main window "JLCPCB Tools" could add "LCSC Package" and better "LCSC Description" from lcsc.com or digikey...

bvernoux, 2021-08-30:
- Main window "JLCPCB Tools" could add "LCSC Package" and "LCSC Description" from https://lcsc.com
  - Maybe first step will be to "parse/extract/display" correctly the "LCSC Description" from the csv as it is not displayed correctly today.
  -  See https://github.com/yaqwsx/jlcparts/blob/master/LCSC-API.md for more details as there is clearly missing details from https://jlcpcb.com/parts and I always check manually the LCSC part on https://lcsc.com to check that the Package, Mfr.Part & Description match with what I want.
  - A must have will be to cross check the Description on LCSC part available on DigiKey as it is more accurate with more details than on LCSC so we will use DigiKey description instead of LCSC...

Bouni, 2021-09-08:
I fixed your first point: "incorrect description" with https://github.com/Bouni/kicad-jlcpcb-tools/commit/0ce0c3d77250125afff72d7118a4019c41b81ee9
The encoding of the CSV is `gbk` rather than `iso-8859-1`

Bouni, 2021-09-08:
I'm not sure how to iclude the better data from LCSC in a proper way 🤔 
Maybe we add an info button to each row that fetchs the data from LCSC API by the LCSC number.
I did a quick example here: https://gist.github.com/Bouni/3d79ddbd68b8fccceb196ad5c9598f1f
It returns a huge amount of data for a single part, so we might show that as another modal!?

bvernoux, 2021-09-08:
After check of https://gist.github.com/Bouni/3d79ddbd68b8fccceb196ad5c9598f1f
I think it will be perfect to concatenate (with space between param and at end "," when there is an other item after) paramVOList => "paramNameEn" {space} "paramValueEn" {,}
- For that example you could format it like that:
  - Power(Watts) 0.25W, Tolerance ±1%, Temperature Coefficient ±100ppm/℃, Resistance 1kΩ

Bouni, 2021-09-08:
The question is when do we query the API?
If you search for let say "10k", the search yields 625 resuts from the CSV, which means we have to do 625 API calls.
This is far from ideal and LCSC might block the API for such a behaviour.

bvernoux, 2021-09-08:
The best will be to query those API with a special "refresh button" for example and store that in the part_assignments.csv
What do you think about that ?

Bouni, 2021-09-08:
You mean just query for already selected parts?
That might be an option.
Or even better, when the user clicks the select part button on the library dialog, the additional info is fetched from the API and stored in the part_assignments.csv as well. That way we only query when needed and do not re-query for already selected parts.

And maybe add a query details button for the library that triggers the query and spawns a new modal with picture and detailed infos!?

bvernoux, 2021-09-08:
1) The best is your proposal:
> when the user clicks the select part button on the library dialog, the additional info is fetched from the API and stored in the part_assignments.csv as well. That way we only query when needed and do not re-query for already selected parts.

2) It will be amazing to have 
> a query details button for the library that triggers the query and spawns a new modal with picture and detailed infos!?

- Like that we will have all in one to choose correctly the good parts and we will do not need anymore to use a web browser to check the different parts details ...

Bouni, 2021-09-08:
Ok, then lets go for that approach!

Bouni, 2021-09-08:
I've implemented a first version, the result is not yet persisted but the modal works already:

[image]

Is that about what you wanted it to look like?

bvernoux, 2021-09-08:
Amazing !! it work fine on Windows10 tested with latest KiCad Nightly (kicad-r24113 from today)

- Could you add also the "Show part details" button in main Window (JLCPCB Tools) on a selection which have a LCSC part of course ?
  - It will be amazing to do a review of parts with all possible details on part (maybe it will be not required when we will have additional details from LCSC but it is nice to check the part like that with the photo and additional details ...)

Bouni, 2021-09-08:
Yes, sure. Just for your information, there's a bug in the info gathering for some of the parts which I need to fix in order to get it working for all the parts.

Bouni, 2021-09-16:
OK, found the problem, added a button to the main window as well that lets you show the part details for already selected parts.
I concider this issue solved 😄

closed 2021-09-16 by Bouni