Bouni/kicad-jlcpcb-tools #17: "JLCPCB Library" shall display a better Description

bvernoux, 2021-08-30:
In "JLCPCB Library" display add the Description from https://github.com/yaqwsx/jlcparts/blob/master/LCSC-API.md (instead of the ones from https://jlcpcb.com/parts) as the one from https://jlcpcb.com/parts is unreadable and not accurate
- Maybe first step will be to "parse/extract/display" correctly the "LCSC Description" from the csv as it is not displayed correctly today.

Bouni, 2021-09-08:
I fixed "incorrect description" with https://github.com/Bouni/kicad-jlcpcb-tools/commit/0ce0c3d77250125afff72d7118a4019c41b81ee9
The encoding of the CSV is `gbk` rather than `iso-8859-1`

Bouni, 2021-09-08:
As this is kind of the same issue / enhancement as #16 I'll go to work out a solution over there and use it in bth, the main window and the library.

closed 2021-09-08 by Bouni

bvernoux, 2021-09-16:
I confirm all work fine with Windows10 + KiCad Nightly (r24220)