Bouni/kicad-jlcpcb-tools #18: Add a way to remove LCSC reference from the JLCPCB Tools plugin (from the part_assignments.csv file)

bvernoux, 2021-08-30:
Add a way to remove LCSC reference from the JLCPCB Tools plugin (from the part_assignments.csv file)
- In some case it is possible the part_assignments.csv is obsolete and some LCSC reference are wrong and have no replacement (and does not exist in schematic) so today there is no easy way to remove those LCSC reference (except by editing the part_assignments.csv file by hand ...)

Bouni, 2021-09-08:
Done by https://github.com/Bouni/kicad-jlcpcb-tools/commit/93317bfe918a1d6117e2989f03f7561415605edf

Bouni, 2021-09-08:
@bvernoux Please be so kind and test this feature

closed 2021-09-08 by Bouni

bvernoux, 2021-09-08:
The "Remove part" feature work perfectly tested with Windows 10 and KiCad 5.99 Nightly (kicad-r23938)
Perfect job like always !!