Bouni/kicad-jlcpcb-tools #19: Add a way to filter on column (for main window "JLCPCB Tools" and "JLCPCB Library")

bvernoux, 2021-08-30:


Bouni, 2021-09-16:
wx allow to make columns sortable, which I added with the latest commit. Not sure if it sorts the way you want it to but I'm statisfied 😉 
@bvernoux please check if this fit your needs

bvernoux, 2021-09-16:
I confirm the column sort work fine (even if it does not use "natural sort" which means for example when you have C1 to C9 then C10 the natural sort will sort it like that in your case with standard sort it does C10, C1, C2 ....) which is not very convenient but ok

Bouni, 2021-09-17:
I think I could implement a custom sorting function and call it on click to the header. 
But to be honest, thats a lot of work for a minor improvement.
Maybe I go for that in the future and leave it as it is for now.
I hope you can live with that 😅

closed 2021-09-17 by Bouni