Bouni/kicad-jlcpcb-tools #21: Move the JLCPCB/LCSC data parsing(csv) to SQLite database to speedup search/loading ...

bvernoux, 2021-08-30:


Bouni, 2021-08-31:
I'm not convinced that this will bring that much of an improvement. 
We will have to parse and update the database on every refresh of the library which will be very time consuming.

Maybe I build a webservice, kind of a open JLCPCB API that we could query instead of using the CSV at all.
The workload of updateing could be done by the webservice every N minutes and users could just query the API 🤔

bvernoux, 2021-08-31:
Yes so far I think it is not a priority as actual "csv" parsing is quite fast even on modest PC (old CoreI7 ...)
Anyway the webservice seems to be a very good idea for the future.

Bouni, 2021-09-03:
I did a quick test, using an Postgres DB and [SQLmodel](https://sqlmodel.tiangolo.com/) to read the CSV and write the rows into the DB. In my current implementation (maybe not idealy written) on a Hetzner CX11 instance (maybe to small) at 100% CPU, it takes approximately 17 seconds to process 1000 rows. 
That means for the current 137424 rows in the CSV I need roughly 40 Minutes to write that into the database.

I have seperate tables at the moment for manufacturers, packages and so on which might cost more time to process that if I would simply write all data into a flat table, on the other hand I'm not sure what that would mean for the query performance for the future API 🤷🏽 

maybe I'll do a second Test with a flat table to see how long that would take to write all data.

Bouni, 2021-09-03:
I just let it run until it finished and it took 3454 seconds (57 minutes).
My next approach will be async, I hope that that will spedd up things!

Bouni, 2021-09-03:
The async approach took me 3052 seconds. Not that much of a speedup 😞

Bouni, 2021-09-03:
I just replaced the CX11 instance for an CPX11 which has two CPU cores and that almost cut the runtime for 1000 parts in half (10.5 seconds per 1000 parts with the async approach).
So it is clearly a CPU issue. 
But I'm not sure how much mony in terms of monthly cost I am willing to spend for the speedup.

The CX11 costs me 4.15€ a month, the CPX11 4.75€ a month, so thats not that much of a difference, but a CPX21 with 3 cores is at 8.21€ a month, so almost doubled.

Bouni, 2021-09-03:
For intrerested readers, heres my two test scripts:

Syncronous version: https://gist.github.com/Bouni/616504817fc984d5d1ca85f63529b494
Asyncronous Version: https://gist.github.com/Bouni/6abb2137aa36bc26de22f37fef22de0e

Bouni, 2021-09-06:
Next test: On an CPX11 instance without relationships in the db and a syncronous approach I'm down to ~5 seconds for 1000 rows. Again almost doubled. I'm not sure what that means for query performance 🤔

bvernoux, 2021-09-07:
The aim was not you pay a server for that so I think the csv is a very good alternative so far.
- I hate to depend on remote server (called cloud today) especially if you shall pay for that, like we can see with KiCost as when it is broken all users are locked and the plugin is not usable anymore ...

Bouni, 2021-09-08:
I agree, that single point of failure is something that we should avoid.
On the other hand if the search is that much faster, it might be worth it.
Doing the CSV to SQL conversion locally is not worth it in my opinion, even if you have a fast workstation and the conversion is done in lets say 10 minutes, who wants to wait that long every time a new lib version is available?

Going with CSV is ok in my opinion, its much faster that the XLS file JLC provided in the past and the download time and pandas load time is just a few seconds. Even teh search is not that slow in my opinion.

I think we should focus on getting better data from LCSC (#16) and use just the part number, stock and price info from JLC.

Bouni, 2021-09-08:
I'll close this and maybe re-open if we decide later to follow this approach.

closed 2021-09-08 by Bouni