Bouni/kicad-jlcpcb-tools #22: BOM isn´t generated and CPL is empty

radontec, 2021-08-31:
unfortunately, no CPLs are generated. the BOM is also missing. I suspect that the fault lies with the solder bridges. I can not unselect these from the BOM and the CPL.
[image]
[image]

it´s not possible to switch from yes to no
[image]

Bouni, 2021-08-31:
Do you get any error message when clicking the Generate fabrication files button?
I can confirm that it is not possible to toggle the CPL / BOM state for footprints that have just a `REF**` as Reference designator because I look them up by the reference and here its multiple that share the same reference.
I need to look for away to solve this.

radontec, 2021-08-31:
no error message:

2021.08.31 12:19:59 - INFO - Load corrections from https://raw.githubusercontent.com/matthewlai/JLCKicadTools/master/jlc_kicad_tools/cpl_rotations_db.csv
2021.08.31 12:19:59 - DEBUG - Starting new HTTPS connection (1): raw.githubusercontent.com:443
2021.08.31 12:20:00 - DEBUG - https://raw.githubusercontent.com:443 "GET /matthewlai/JLCKicadTools/master/jlc_kicad_tools/cpl_rotations_db.csv HTTP/1.1" 200 553
2021.08.31 12:20:02 - INFO - Loaded Library with 227970 parts
2021.08.31 12:20:05 - INFO - Successfully ploted Top layer
2021.08.31 12:20:05 - INFO - Successfully ploted Silk top
2021.08.31 12:20:05 - INFO - Successfully ploted Mask top
2021.08.31 12:20:05 - INFO - Successfully ploted Bottom layer
2021.08.31 12:20:05 - INFO - Successfully ploted Silk top
2021.08.31 12:20:05 - INFO - Successfully ploted Mask bottom
2021.08.31 12:20:05 - INFO - Successfully ploted Edges
2021.08.31 12:20:05 - INFO - Finished generating Excellon files
2021.08.31 12:20:05 - INFO - Finished generating ZIP file

Bouni, 2021-08-31:
I thought a bit about this issue and I will simply ignore footprints that have no reference (REF**).
Its very simple to implement and those are invalid for production anyway.

Bouni, 2021-08-31:
@radontec Would you please test my latest commit? 
Works for me:

[image]

radontec, 2021-08-31:
got a new exception
[image]

Bouni, 2021-09-01:
@radontec Fixed this issue in the last commit, would you mind trying again?

radontec, 2021-09-01:
now it works perfect! love this plugin

closed 2021-09-01 by radontec