Bouni/kicad-jlcpcb-tools #24: Error installing plugin?

mrwilby, 2021-09-16:
I wasn't able to see the plugin (it's installed in the correct directory). Running on a mac, using nightly: kicad-unified-20210915-042000-ca6e43d1e1.dmg

```
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/me/Documents/KiCad/5.99/scripting/plugins/kicad-jlcpcb-tools/__init__.py", line 4, in <module>
    from .plugin import JLCPCBPlugin
  File "/Users/me/Documents/KiCad/5.99/scripting/plugins/kicad-jlcpcb-tools/plugin.py", line 28, in <module>
    from .library import JLCPCBLibrary
  File "/Users/me/Documents/KiCad/5.99/scripting/plugins/kicad-jlcpcb-tools/library.py", line 20, in <module>
    subprocess.check_call(["python", "-m", "pip", "install", "pandas"])
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['python', '-m', 'pip', 'install', 'pandas']' returned non-zero exit status 1.
```

Bouni, 2021-09-17:
The problem is clear to me, the plugin installs the python pandas package by [calling pip](https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/library.py#L14-L21)
This works fine on Windows and linux but seems not to work on MacOS.
Unfortunately I have no Mac to test.

You could test something for me in order to fix this. Open your board file then open the Python Scripting console

[image]

In the console type `import pip` and hit Enter, no error should appear. If thats the case, I thin I have a solution for MacOS as well 😄

mrwilby, 2021-09-17:
Thank you for your help! I can test, no worries.

```import pip```

...was ok, no error.

Bouni, 2021-09-17:
My idea didn't work out 😞 

But you should be able to install pandas by hand, just open a terminal an run `python -m pip install pandas`

I'm actually not sure if KiCAD for MacOS brings its own python runtime or depends on the system version like linux does 🤔

mrwilby, 2021-09-17:
Thank you. On my mac installation of kicad, python appears to be installed here:

```/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.8```

If I try to manually install the module, I see an error:

```❯ python -m pip install pandas
Could not import runpy module```

I didn't find a solution yet...

mrwilby, 2021-09-17:
Well, I did figure out how to pip install pandas, but then that resulted in a new error:

```
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pandas/__init__.py", line 16, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.8 from "/Applications/KiCad/KiCad.app/Contents/MacOS/kicad"
  * The NumPy version is: "1.21.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so, 2): no suitable image found.  Did find:
	/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so: code signature in (/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so) not valid for use in process using Library Validation: mapped file has no Team ID and is not a platform binary (signed with custom identity or adhoc?)

During handling of the above exception, another exception occurred:

**********************************
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/me/Documents/KiCad/5.99/scripting/plugins/kicad-jlcpcb-tools/__init__.py", line 4, in <module>
    from .plugin import JLCPCBPlugin
```

DanielO, 2021-10-08:
I had this problem too and I don't think it is solvable in MacOS.
I ended up forking the code and using sqlite: https://github.com/DanielO/JLCKicadTools (sqlite branch)
It is based on a fork (I think?) of this code though.

It is a bit horrible because it spawns iconv to convert the CSV to UTF-8 then feeds that to the sqlite3 binary to convert it into a database table.. It probably won't work in Windows (without obtaining iconv and sqlite binaries) but works fine in MacOS and I expect would work fine in Linux.

I also made a few other "quality of life" improvements too.

Bouni, 2021-10-08:
@DanielO I don't have access to a MacOS machine so solving this issue is quite hard for me.
Is sqlite already installed as a python package with KiCAD?

> It is based on a fork (I think?) of this code though.
No, I just use the rotation info used by matthewlai's
JLCKicadTools. See https://github.com/Bouni/kicad-jlcpcb-tools/issues/1 for more infos.

As far as I can tell from a quick look at matthewlai's code, you run the script externally from a shell, right?
It'ts not started from within KiCAD as an action plugin!?

Bouni, 2021-10-08:
@DanielO Just to get my head around it, you forked my plgin and replaced the pandas parts with sqlite, right?
I'm confused as well :zany_face:

I did some SQL tests myself and for me the time needed to get the CSV data into the database was not worth it (See https://github.com/Bouni/kicad-jlcpcb-tools/issues/21)

On Windows, KiCAD ships its own python environment including the interpreter, so thats why I built the pandas install into the plugin https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/library.py#L13-L21
On Linux it uses the systems python installation (/usr/bin/python3 in my case)

Would you mind helping me out and opening the Python console from within your KiCAD installation and running 
`import sys` followed by `print(sys.executable)`?
That way I would know what python installtion is used on MacOS and probably find a solution for the pandas issue.

DanielO, 2021-10-08:
> So it's kind of a Frankenstein Plugin now 😆
Wow what a mess, well I can fix that on the weekend I guess!

DanielO, 2021-10-08:
> And I've seen that you don't use the python sqlite bindings but rather call the sqlite3 comandline utility via subprocess.
> Ok, but that way you need to rely on the user having that installed. I would prefer a way of installing a python package from within the plugin the create a smooth experiance for the user 😄

I think it is better if the plugin pops up a dialog saying what the user needs to install, possibly with a "do it for me" button.

I am honestly surprised it works on Linux - I would have expected permission denied errors as pip tries to write to /usr/lib/python3.9/site-packages and the like.

Bouni, 2021-10-08:
If you create a fork with a sqlite branch I'll happily test it out on linux and windows and if I'm statisfied I'll for sure merge it into master. What I did withmy SQL tests was read line by line and write that into the db which is hard work for the machine but if we can get this down to a few seconds that could be totally worth it!

On linux the package is installed into the users home, not the site-packages of the system.