Bouni/kicad-jlcpcb-tools #72: Submit your plugin to the PCM metadata repo

nickoe, 2021-11-09:
I just found this tool from this tweet
https://twitter.com/arturo182/status/1451527309907927057

You may be interested in adding support for the new Plugin And Content Manger (PCM) to make it easier for users to install. The PCM feature is now enabled by default in the nightlies.

Documentation on this is available in:
https://dev-docs.kicad.org/en/addons/

Bouni, 2021-11-10:
I thought exactly that after reading @arturo182 's [tweet](https://twitter.com/arturo182/status/1458201297790832645) yesterday 😄 
I'll defenitely do that, but that will take a little more time in order to get things to comply with their rules.

Bouni, 2021-12-08:
I currently work on a GitHub action that generates the needed files for submission to the PCM repo every time I create a release.
Once that works I'll submit to PCM 🤓

Bouni, 2021-12-08:
Lets see if this succeeds: https://gitlab.com/kicad/addons/metadata/-/merge_requests/22

At the moment when I create a new release the files are generated automatically and added as release assets but I still have to manually create a Merge Request over at GitLab. Maybe I can automate that as well in the future ....

Bouni, 2021-12-09:
😢 
I tottaly overlooked this section of the dev docs on PC addons:

> Commercial Services
>
> Packages that link to or provide commercial services, including but not limited to PCB fabrication, component lookup and order management, must first contact the KiCad team at plugins@kicad.org to discuss commercial plugin options.

So this plugin will never be in the **official** PCM list (I doubt JLC will do anything to convince KiCAD to do that), **but** I'm going to create a custom repo that can be added to the PCM list of repos which will make the plugin installabel from there.

Bouni, 2021-12-09:
🥳 

This not yet fully automated but it works for now!

[image]

Add my addon repo and you can install the plugin via the Plugin Manager 😁

AiGreek, 2021-12-09:
> So this plugin will never be in the **official** PCM list (I doubt JLC will do anything to convince KiCAD to do that), **but** I'm going to create a custom repo that can be added to the PCM list of repos which will make the plugin installabel from there.

Are they **** serious ?? Just because it's for JLCPCB you can't publish it ? *sigh*

Bouni, 2021-12-09:
That's how it is 🤷

Anyway, I think my custom repo is a good alternative.

Bouni, 2021-12-10:
🥳 

When ever I create a new release, a ZIP file is generated using [GitHub actions](https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/.github/workflows/kicad-pcm.yml) and uploaded as a release asset.
Then the regeneration, also a [GitHub action](https://github.com/Bouni/bouni-kicad-repository/blob/main/.github/workflows/rebuild.yml) of the custom addon repo is trigged via a `workflow_dispatch`.
That keeps the repo up-to-date automatically! The only downside is the 5 minute cache timeout on those files.
If you try to load the addon in this period you'll most likely get an error that the sha256 sums are not correct.

All in all I can live with this solution 😁

mikwielgus, 2021-12-13:
@Bouni Are you confident that JLCPCB is not going to contact the KiCad team to approve your plugin if asked?

As far as I know, the plugin doesn't have to originate from the commercial entity it interacts with (i.e. it can be third party), but we just need to obtain an explicit approval from that entity. I've asked Seth to clarify what he wrote in your merge request.

The wording in the developer docs should be clarified, as it clearly has caused some confusion. And I definitely want your plugin to be available in the official repository.

Bouni, 2021-12-13:
@mikwielgus I don't know but so far JLC never answerd a single of my e-mails with inquiries (see #55 ).
They promised in the live chat to reach out to me but nothing happened so far.

But if you guys clearify if a simple "go for it" form the comercial entity is enough, I'll try to get JLC to send that e-mail 😄 

> And I definitely want your plugin to be available in the official repository.

Thanks for your support :relaxed:

mikwielgus, 2021-12-13:
Just to be clear, it's more than just a simple "go for it" -- apparently a contract is necessary as it stands currently (I don't yet know exactly why) -- but this obviously also can be done. It's well within JLCPCB's interest to get your plugin in the official repository.

Please keep me updated with your progress if you intend to go with this. I'll try to help if there are some obstacles.

Bouni, 2021-12-13:
Ok, thats about what I expected. 
Is "apparently a contract is necessary as it stands currently (I don't yet know exactly why)" the answer you've got from Seth?

I'll see if JLCPCB is willing to do whats neccessary but before I ask them it would be nice to have a clear statement what exactly is needed in order to get permission.

yaqwsx, 2021-12-13:
@Bouni: See the discussion on the KiCAD's Discord server in the channel scripting (about two days ago). I was discussing it with Seth and Jon - there is a useful insight and explanation.

Bouni, 2021-12-13:
You guys are just awesome! Thanks for all your efforts getting this into the offical PCM repo 😄

Bouni, 2021-12-14:
Just for documentation purposes, @yaqwsx asked @JLCPCB on Twitter for help with this issue and they said the woul reach out to me later (Whenever "later" might be 😏 ) https://twitter.com/JLCPCB/status/1470305011846762496

OmegaRogue, 2021-12-15:
I think some sort of unofficial or community PCM repo would be useful for this sort of thing

OmegaRogue, 2021-12-16:
> I think some sort of unofficial or community PCM repo would be useful for this sort of thing

https://github.com/OmegaRogue/Unofficial-KiCad-Plugins I'm setting one up, the github actions workflow isn't working yet, but it's essentially just one script that gets executed

Bouni, 2021-12-24:
I was in touch with JLCPCB via Twitter private messages after I've [posted this](https://twitter.com/bouni2016/status/1473570221294239746). After Christmas I should get a reply on this topic 🥳

OmegaRogue, 2022-01-19:
Any update on this?

Bouni, 2022-01-19:
Nope, a lot of promises from JLC but nothing else so far ...

Bouni, 2022-02-15:
I've finally got an email from JLCPCB but they did not say anything about the PCM issue. 
I answered with another request to send that email to the KiCAD addon team.

We'll see if the do it or not (or let us wait another 3 months ... 😏 )

closed 2022-06-08 by Bouni