Bouni/kicad-jlcpcb-tools #75: Plugin not detected latest Nightly

AiGreek, 2021-11-14:
Since the last update of Kicad Nightly, the plugin is no longer detected by the software (the icon no longer appears although the plugin is present in the corresponding folder) but the other plugins seem to work (KiBuzzard)

I downloaded the latest "main" version

```
Application: KiCad PCB Editor (64-bit)

Version: (5.99.0-13263-g43c85771eb), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Nov 13 2021 00:01:08
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2021-11-15:
I just installed the latest nightly

```
Application: KiCad PCB Editor (64-bit)

Version: (5.99.0-13288-g9941a52e75), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Nov 14 2021 12:01:02
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

It seems that python `requests` is no longer shipped with the latest nightly builds for some reason 😞 
I'll try to make the plugin install it automatically and hope it works on all platforms.
I did that in prior versions with `pandas` which didn't work on OSX because that required the building of some c extensions ...

We'll see 😅

Bouni, 2021-11-15:
I fixed this issue with 44c86dcdaa6ee18349fa762936b6cb5907569efa but it kind of hacky to install dependencies this way.
Also I'm not sure if this works with OSX 🤔 
@DanielO Maybe you can verify if this works as you pointed out the last time that it didn't for pandas but reuests is pure python as far as I know, so this could work on OSX as well ...

DanielO, 2021-11-15:
Blergh  :(

I downloaded kicad-unified-20211114-050659-f5fc9fa11f.dmg (the latest) and it has requests in it at `/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests`
(I also checked and the plugin works)

It seems very odd Windows wouldn't ship with it when OSX does - perhaps it was an oversight, I think it would be worth filing a bug about it.

Bouni, 2021-11-15:
Thanks @DanielO for again helping out with OSX testing 😃 
I guess you're right!

Interestingly there is a version of request installed, but as a pip sub module:

`C:\Program Files\KiCad\5.99\bin\Lib\site-packages\pip\_vendor\requests` and I'm not able to import that one.

I also just verified with a version from october and indeed back then was a version of requests packaged!

`C:\Program Files\KiCad\5.99\bin\Lib\site-packages\requests`

DanielO, 2021-11-15:
I think it would be definitely worth raising it as an issue with the KiCAD team, obviously the exact mix of packages is dependent on what KiCAD needs but it does seem like a strange thing that it's on the OSX build but not the Windows one.

Bouni, 2021-11-15:
I've already done that!

https://gitlab.com/kicad/code/kicad/-/issues/9644

Bouni, 2021-11-15:
Just verified that this is fixed in the latest nightly build `kicad-msvc.r25153.2340f82091`

closed 2021-11-15 by Bouni