Bouni/kicad-jlcpcb-tools #76: Can't generate files in "major-refractor"

AiGreek, 2021-11-20:
I wanted to test the _"major-refractor"_ branch (I love the addition of the icons <3) 
except that when I click to _Generate the GERBER_ nothing happens.

```
Application: Editeur de PCB KiCad (64-bit)

Version: (6.0.0-rc1-1-g4df25edb3d), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Nov 16 2021 00:01:03
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```
A question that I also ask myself. I know the French are lazy and a bit ..... well .... let's just say lazy ^^. 
Do you think that if I offer an update to translate the plugin would be useful ?

Bouni, 2021-11-21:
Hi, the refactor branch is not yet ready to use. I estimate one more week and it should be ready for test.

I'll let you know in this issue when the time has come 😉

To be honest, I need to look into how translations can be made possible in wx but if its not that hard I defenitely  going to do that!

AiGreek, 2021-11-22:
> Hi, the refactor branch is not yet ready to use. I estimate one more week and it should be ready for test.

Seems quite ready for me except gerber generation and prices (but i don't need it so .... :p)
very awesome job on this plugin. I translated it a few days ago but i have to re-do it each time i upload it ^^

Bouni, 2021-11-23:
@AiGreek I think its worth a try now but expect some issues here and there, I've had no time to properly test everything

AiGreek, 2021-11-23:
Seems to work perfectly. I have no idea how to use the rotations part, but everything is ok for me

Bouni, 2021-11-23:
JLC has different part orientations than KiCad has.
Lets say a SOT-23 package has its single leg facing north in KiCad, resulting in a rotation of 0° but JLC expects the same orientation to be 90°.
Then you can add an entry to the rotations db with a regex `^SOT-23` and a correction of 90°.
When you then generate the POS file, all packages are checked if they match any of the regex in the db and if so, the correction is added.
That way you get fitting POS files for JLC.

closed 2021-12-04 by AiGreek