Bouni/kicad-jlcpcb-tools #78: Download of the CSV failed, no existing library found, exit plugin now

katodo, 2021-12-03:
I just installed this plugin but I've this error on opening:
"Download of the CSV failed, no existing library found, exit plugin now"

[image]

am I forgetting some initial steps?

Bouni, 2021-12-03:
This is because JLCPCB from time to time fails at serving there parts database CSV.
Try it later, maybe they have fixed it then.

I guess I have to setup my own caching server to avoid this problem bin the future 😑

pixsysedp, 2021-12-03:
I'm testing now from another network on a windows pc but there are same problem... can you exchange CSV here?

Bouni, 2021-12-03:
Unfortunately not at the moment.
The JLCPCB link gives a 404 😕

obra, 2021-12-04:
Would it make sense to seed the parts database with an ancient copy in git? Alternatively, is it possible to make the fabrication outputs functionality not depend on the download?

AiGreek, 2021-12-04:
Same error for me. LCSC seems to be unavailable. If it can help i uploaded  the two files needed to use the plugin.

**_KiCad\6.0\scripting\plugins\JLCPCB_**
```
https://www.dropbox.com/s/l68gpu5s434zrcj/jlcpcb_parts.db?dl=0
https://www.dropbox.com/s/vqfrrwnyccee2ea/parts.db?dl=0
```

Bouni, 2021-12-04:
> Would it make sense to seed the parts database with an ancient copy in git? Alternatively, is it possible to make the fabrication outputs functionality not depend on the download?

I think caching the CSV is the better option.
The generation without the possibility to select and assign LCSC part numbers seems kind of useless to me, why would you do that?

Bouni, 2021-12-04:
> Same error for me. LCSC seems to be unavailable. If it can help i uploaded  the two files needed to use the plugin.
> 
> 
> **_KiCad\6.0\scripting\plugins\kicad-jlcpcb-tools\jlcpcb_**
> ```
> https://www.dropbox.com/s/l68gpu5s434zrcj/jlcpcb_parts.db?dl=0
> https://www.dropbox.com/s/vqfrrwnyccee2ea/parts.db?dl=0
> ```

The parts.db is not needed, it's project specific!

AiGreek, 2021-12-04:
woops my bad ^^'

obra, 2021-12-04:
In this case, I already have manually assigned LCSC parts on most of my BOM.  But also, their web based part assignment doesn’t suck.

Bouni, 2021-12-04:
> In this case, I already have manually assigned LCSC parts on most of my BOM.  But also, their web based part assignment doesn’t suck. 

Ok, I've not seen that aspect so far.
I'll look into it next week after my vacation

obra, 2021-12-04:
Have a great vacation!

Bouni, 2021-12-06:
I totally forgot that this is the default behaviour in the now merged `major-refactor` branch.
You get an error but still can generate the files

closed 2021-12-06 by Bouni