Bouni/kicad-jlcpcb-tools #80: Add import/export capability for footprint rotation corrections

Bouni, 2021-12-08:
We need a way to exprot/import footprint rotation corrections in the rotation manager.
ideally the resulting format is a CSV that is compatible with that used by @matthewlai in https://github.com/matthewlai/JLCKicadTools

atlaste, 2022-01-24:
First off, great plugin!

I think this is not just limited to rotations, but should include offsets as well. 12251103CNG0S115001 is an example part number (standard 2.54mm pin headers). 

I also noticed it outputs values like "-90.0", which should be "270.0" for rotation. Tested in Kicad 6 which is unsupported but seems to work fine nonetheless ;-)

atlaste, 2022-01-25:
Allright, on further investigation, this seems to be about through hole parts. Apparently, through hole parts are not centered in KiCad, but rather are centered on pin 1. I'm not sure who came up with that idea.

Regardless, if there is some way to get the offset of pin 1, it should be possible to compensate for that. The only downside is that it doesn't hold for all the parts with through holes... F.ex. I have a micro-SD connector, which has tiny mounting holes that appear in the footprint as through holes. 

I also see small offset differences in the preview window of jlc for bigger boards the more I go to the bottom right, but i'm currently unsure if that's because of jlc or because of the tool... will have to investigate that a bit further.

corlconf, 2022-01-26:
Yeah through-hole part offsets are a mess in general. Also modules like the ESP32. I don't think there is really a standard. Personally I just submit it and let JLC deal with it, and that has always worked for me.

matthewlai, 2022-01-26:
Yeah through-hole part offsets are a mess in general. Also modules like the ESP32. I don't think there is really a standard. Personally I just submit it and let JLC deal with it, and that has always worked for me.

Bouni, 2022-01-26:
I could for sure add corrections offsets in X and Y in addition to the rotation correction, thats not a big deal! 🤔

atlaste, 2022-01-26:
> I also see small offset differences in the preview window of jlc for bigger boards the more I go to the bottom right, but i'm currently unsure if that's because of jlc or because of the tool... will have to investigate that a bit further.

Further investigation tells me that's just JLC messing up.

Breakout modules like the ESP32-DevkitC are a total mess. What you would ideally want is to just have 2 19x1 pin headers. Currently I would have to overlay them I suppose.

Bouni, 2022-01-26:
> Breakout modules like the ESP32-DevkitC are a total mess. What you would ideally want is to just have 2 19x1 pin headers. 

You mean to have a ESP32-DevkitC mounted to the PCB you want to order you need to populate 2 seperate rows of female headers, right?

I think there's no other way than having two headers in you schematic, J1 and J2 for example, assign each the same LCSC number and place them correctly on your board.

atlaste, 2022-01-26:
Correct.

> I could for sure add corrections offsets in X and Y in addition to the rotation correction, thats not a big deal!

Yes please. Please first correct and then rotate; that way we can use the measure functionality in the footprint for corrections, instead of having to think about rotations...

Additionally, it would be great if it were possible to calculate the center point from the pins, with a checkbox... It's not a general rule, but for lots of components (especially with through holes) simply averaging the pin centers would work quite well I think. (Think about pin headers, terminals, IC's, etc).

Other than that your tool works pretty good.

Bouni, 2022-01-27:
@atlaste I totally agree with all the above. 
I think I should try to add kind of a graphical preview of the corrections, for example show the original position slightly transparent, overlay it with the corrected footprint and update that when ever the settings are changed.
What do you think?

atlaste, 2022-01-27:
Well if that's possible, go for it. 
You might even want to use the center point [of the pins] as the starting point instead of the kicad center (after all, JLC calls it the "centre point"). I'm really not that good with Python, but I'm happy to test it with some designs here to see how good it works. If it works great, there's no need to do anything manually...

Bouni, 2022-01-28:
@atlaste showing the footprint turns out to be harder than I initially thought but I can eventually get this working but it takes some time.

closed 2022-02-02 by Bouni