Bouni/kicad-jlcpcb-tools #83: KiCad freezes when clicking on jlc button

chevroux, 2021-12-10:
Hey there!

When I click on the jlc-button in pcbnew the program unfortunatly just freezes. I tried to wait a few minutes without luck :(
A jlcpcb-Folder gets created inside my project folder, it has an empty assembly and gerber folder. And a project.db listing all my parts.

I'm on KiCad 6.0.0-rc1-160-g99ec2b1801 on Win10 64bit.

Bouni, 2021-12-10:
Could you please try to open the python sycripting console before you try to start the plugin?

[image]

That will spawn a stdout/stderr window if something goes wrong during the plubin start.

If so, please post the output of that window.

chevroux, 2021-12-10:
Thanks for the quick reply!
This is what I get:
[image]

Unfortunatly I can't resize or scroll the window since kicad and the window aren't responsive anymore.

Bouni, 2021-12-10:
Strange, let me update my KiCAD to the exact same version as you have in order to see if I can reproduce it.
Can you share your project (as a zip) so that I can test if its something project related?

chevroux, 2021-12-10:
Hey again,

Sure, here it is:
[jlc_prob.zip](https://github.com/Bouni/kicad-jlcpcb-tools/files/7692604/jlc_prob.zip)

Bouni, 2021-12-10:
I've just updated to the latest version becaus I was not able to figure out which version you have (6.0.0-rc1-160-g99ec2b1801 does not match any of the available downloads).

PCB Editor -> Help -> Abut KiCAD -> Copy Version Info givs me now 
```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.0-rc1-392-g84e0feeee6), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 10 2021 09:00:19
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```
Please post that version info as well. I guess that your version is quite old as `-160-` is way behind `-392-`.

I open your project and click the icon and it works as expected:

[image]

chevroux, 2021-12-10:
Yeah, mine is a bit old:

Application: KiCad (32-bit)

Version: (6.0.0-rc1-160-g99ec2b1801), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19042), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Nov 26 2021 00:03:53
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

I'll update to the newest nightly and report back if anything changes

chevroux, 2021-12-10:
Unfortunatly got the same behavior with the latest version:

`Application: KiCad (64-bit)

Version: (6.0.0-rc1-392-g84e0feeee6), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19042), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 10 2021 09:00:19
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
`

Bouni, 2021-12-10:
How did you install the plugin?
And into which folder?

chevroux, 2021-12-10:
I just downloaded the Code with Code -> Download ZIP and extracted it in the plugins folder:
[image]

Bouni, 2021-12-10:
To be honest, I have not the slightest clue what goes wrong here :worried:
Let me finish the automatic release builder and custom addon repo. After that you can try installing the plugin via the Plugin Manager. Hopefully that will work!

chevroux, 2021-12-10:
No prob, thanks for your time! One last suggestion: could my locale be the problem? My OS in in German.

Bouni, 2021-12-10:
I dont think so as mine is German as well 😉

chevroux, 2021-12-10:
Ah, okay :D Thanks again, I'll close the issue and try my luck with the plugin manager in the future =)

closed 2021-12-10 by chevroux