Bouni/kicad-jlcpcb-tools #87: Exception when trying to use the tool

nickoe, 2021-12-16:
I get the following exception, not sure what to make of it:
[image]

I also recently updated my system, including a ["rebuild"](https://github.com/archlinux/svntogit-community/commits/packages/python-wxpython/trunk) of wxpython
```
[2021-12-15T20:49:56+0100] [ALPM] upgraded python-wxpython (1:4.0.7.2-1 -> 1:4.0.7.2-3)
```

Kicad version info.
```
Application: KiCad PCB Editor

Version: (6.0.0-rc2), release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.80.0 OpenSSL/1.1.1l zlib/1.2.11 brotli/1.0.9 zstd/1.5.0 libidn2/2.3.2 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.10.0 nghttp2/1.46.0

Platform: Linux 5.15.6-arch2-1 x86_64, 64 bit, Little endian, wxGTK, i3, x11

Build Info:
	Date: Dec 16 2021 15:43:50
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.76.0
	OCC: 7.5.3
	Curl: 7.80.0
	ngspice: 35
	Compiler: Clang 13.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2021-12-16:
I think I remember having the same (or at least similar) problem on my Arch notebook when ever I've upgraded my system which upgraded wxpython to a version greater than 4.0.7.something 🤔 

Let me double check that. Maybe you can try downgrading that and giving it a try.

Bouni, 2021-12-16:
I can confirm that I see the same problem on my Arch notebook but downgrading wx didn't help.
I'll try to figure out how to get this working again.

Bouni, 2021-12-17:
I found a potential solution for this problem, but I'll wait for @DanielO to test this on OSX to prevent creating a regression there before merging this into main.

Bouni, 2021-12-20:
I'm going to merge this as its a show stopper for linux users. If this is a regression on OSX I'll have look into it when somebody creates an issue about it.

closed 2021-12-20 by Bouni