Bouni/kicad-jlcpcb-tools #91: Crash after download on OSX

DanielO, 2021-12-20:
I updated to c59a983 but it crashed after downloading the library.
I think it is most likely due to a GUI call being made on a non-main thread but I have not investigated yet.

I rolled back to abf2147ba2a343da5d45342e4e5760977502a0a2 and it works again.

Here's the (trimmed) crash dump:

```
Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [83430]

Application Specific Information:
BUG IN CLIENT OF LIBDISPATCH: Assertion failed: Block was expected to execute on queue [com.apple.main-thread]

Thread 0:: Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	0x00007fff2020b6e0 objc_retain + 80
1   com.apple.AppKit              	0x00007fff23509e6b -[NSView _updateTrackingAreasWithInvalidCursorRects:] + 2057
2   com.apple.AppKit              	0x00007fff22daf9a5 -[NSScrollView _updateTrackingAreasWithInvalidCursorRects:] + 134
3   com.apple.AppKit              	0x00007fff23509f10 -[NSView _updateTrackingAreasWithInvalidCursorRects:] + 2222
4   com.apple.AppKit              	0x00007fff23509f10 -[NSView _updateTrackingAreasWithInvalidCursorRects:] + 2222
5   com.apple.AppKit              	0x00007fff22d4c7a1 _NSWindowDisplayCycleUpdateStructuralRegions + 212
6   com.apple.AppKit              	0x00007fff22d4760f __NSWindowGetDisplayCycleObserverForUpdateStructuralRegions_block_invoke + 420
7   com.apple.AppKit              	0x00007fff22d40ea5 NSDisplayCycleObserverInvoke + 155
8   com.apple.AppKit              	0x00007fff22d40a30 NSDisplayCycleFlush + 953
9   com.apple.QuartzCore          	0x00007fff26bd6c86 CA::Transaction::run_commit_handlers(CATransactionPhase) + 92
10  com.apple.QuartzCore          	0x00007fff26bd5a1d CA::Transaction::commit() + 375
11  com.apple.AppKit              	0x00007fff22df086c __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
12  com.apple.AppKit              	0x00007fff23546332 ___NSRunLoopObserverCreateWithHandler_block_invoke + 41
13  com.apple.CoreFoundation      	0x00007fff20464671 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
14  com.apple.CoreFoundation      	0x00007fff20464505 __CFRunLoopDoObservers + 543
15  com.apple.CoreFoundation      	0x00007fff20463998 __CFRunLoopRun + 841
16  com.apple.CoreFoundation      	0x00007fff20462f8c CFRunLoopRunSpecific + 563
17  com.apple.HIToolbox           	0x00007fff286ab1f3 RunCurrentEventLoopInMode + 292
18  com.apple.HIToolbox           	0x00007fff286aae26 ReceiveNextEventCommon + 284
19  com.apple.HIToolbox           	0x00007fff286aacf3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
20  com.apple.AppKit              	0x00007fff22c6c172 _DPSNextEvent + 864
21  com.apple.AppKit              	0x00007fff22c6a945 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
22  com.apple.AppKit              	0x00007fff22f42ea6 -[NSApplication _doModalLoop:peek:] + 301
23  com.apple.AppKit              	0x00007fff230cd5ee __33-[NSApplication runModalSession:]_block_invoke_2 + 69
24  com.apple.AppKit              	0x00007fff230cd596 __33-[NSApplication runModalSession:]_block_invoke + 78
25  com.apple.AppKit              	0x00007fff22f41470 _NSTryRunModal + 100
26  com.apple.AppKit              	0x00007fff230cd491 -[NSApplication runModalSession:] + 128
27  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b4c3860 wxGUIEventLoop::DoDispatchTimeout(unsigned long) + 64
28  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b3b0563 wxCFEventLoop::DispatchTimeout(unsigned long) + 35
29  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b3b06ed wxCFEventLoop::OSXDoRun() + 189
30  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b4c3f1e wxModalEventLoop::OSXDoRun() + 110
31  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b3b081d wxCFEventLoop::DoRun() + 29
32  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b2f73ce wxEventLoopBase::Run() + 158
33  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b40e3d5 wxDialog::ShowModal() + 101
34  _core.cpython-38-darwin.so    	0x000000012a2484c8 meth_wxDialog_ShowModal(_object*, _object*) + 104
35  org.python.python             	0x000000010c4fedc2 cfunction_call_varargs + 290
36  org.python.python             	0x000000010c4fe785 _PyObject_MakeTpCall + 373
37  org.python.python             	0x000000010c5cced5 call_function + 533
38  org.python.python             	0x000000010c5c9c8d _PyEval_EvalFrameDefault + 25677
39  org.python.python             	0x000000010c4ff0d0 function_code_fastcall + 128
40  org.python.python             	0x000000010c5017db method_vectorcall + 235
41  org.python.python             	0x000000010c4ff9f0 PyObject_CallObject + 80
42  _pcbnew.kiface                	0x00000001260bd7ed PYTHON_ACTION_PLUGIN::CallMethod(char const*, _object*) + 93
43  _pcbnew.kiface                	0x00000001260be166 PYTHON_ACTION_PLUGIN::Run() + 38
44  _pcbnew.kiface                	0x00000001260be629 PCB_EDIT_FRAME::RunActionPlugin(ACTION_PLUGIN*) + 937
45  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b3e18dd wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 301
46  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b3e14f3 wxEvtHandler::ProcessEvent(wxEvent&) + 163
47  _pcbnew.kiface                	0x0000000126559fc7 EDA_BASE_FRAME::ProcessEvent(wxEvent&) + 87
48  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b3e1651 wxEvtHandler::ProcessEventLocally(wxEvent&) + 145
49  libwx_osx_cocoau-3.1.5.0.0.dylib	0x000000010b3e14b0 wxEvtHandler::ProcessEvent(wxEvent&) + 96
[... 78 lines cut ...]
32  org.python.python             	0x000000010c5cdcc4 _PyEval_EvalCodeWithName + 2804
33  org.python.python             	0x000000010c4ff26e _PyFunction_Vectorcall + 270
34  org.python.python             	0x000000010c5cce7c call_function + 444
35  org.python.python             	0x000000010c5c9c69 _PyEval_EvalFrameDefault + 25641
36  org.python.python             	0x000000010c4ff0d0 function_code_fastcall + 128
37  org.python.python             	0x000000010c5cce7c call_function + 444
38  org.python.python             	0x000000010c5c9c69 _PyEval_EvalFrameDefault + 25641
39  org.python.python             	0x000000010c4ff0d0 function_code_fastcall + 128
40  org.python.python             	0x000000010c5017db method_vectorcall + 235
41  org.python.python             	0x000000010c4feaad PyVectorcall_Call + 109
42  org.python.python             	0x000000010c5ca198 _PyEval_EvalFrameDefault + 26968
43  org.python.python             	0x000000010c4ff0d0 function_code_fastcall + 128
44  org.python.python             	0x000000010c5cce7c call_function + 444
45  org.python.python             	0x000000010c5c9c69 _PyEval_EvalFrameDefault + 25641
46  org.python.python             	0x000000010c4ff0d0 function_code_fastcall + 128
47  org.python.python             	0x000000010c5cce7c call_function + 444
48  org.python.python             	0x000000010c5c9c69 _PyEval_EvalFrameDefault + 25641
49  org.python.python             	0x000000010c4ff0d0 function_code_fastcall + 128
50  org.python.python             	0x000000010c5017db method_vectorcall + 235
51  org.python.python             	0x000000010c4feaad PyVectorcall_Call + 109
52  org.python.python             	0x000000010c66c7da t_bootstrap + 74
53  org.python.python             	0x000000010c61fb19 pythread_wrapper + 25
54  libsystem_pthread.dylib       	0x00007fff2036d8fc _pthread_start + 224
55  libsystem_pthread.dylib       	0x00007fff20369443 thread_start + 15
```

Bouni, 2021-12-20:
😭 I feared that this would happen
Anyway thanks for testing on OSX!

I guess there is a difference in the packaged wx / wxpython version on OSX / Linux / Windows.

I think I'll check the OS with `os.platform.system() == "darwin"` and use the prvious version if thats the case and the new one if its linux or windows.
A hacky solution but I have no intention to debug wx to get this up and running on all platforms 😅

DanielO, 2021-12-20:
Yeah fair enough! wx does seem to have some sharp edges..

Bouni, 2021-12-20:
@DanielO Please test my latest commit and let me know if this works now on OSX.

eric-wood, 2021-12-20:
Chiming in that I'm experiencing the same issue and it appears to still be happening for me with that updated branch. Let me know if there's any additional information you'd like! I'd offer to dive deeper but this is a bit out of my wheelhouse.

<details>
<summary>Full crash report</summary>
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               kicad [89250]
Path:                  /Applications/KiCad/KiCad.app/Contents/MacOS/kicad
Identifier:            org.kicad.kicad
Version:               (6.0.0-rc1-249-g84cfa052a1) ((6.0.0-rc1-249-g84cfa052a1))
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2021-12-20 16:34:05.6733 -0700
OS Version:            macOS 12.0.1 (21A559)
Report Version:        12
Bridge OS Version:     6.0 (19P549)
Anonymous UUID:        6A48758C-3622-D0C3-472A-396A60CC89EF

Sleep/Wake UUID:       83B1FD7A-A19F-488E-BE9F-652C45F20674

Time Awake Since Boot: 350000 seconds
Time Since Wake:       1705 seconds

System Integrity Protection: enabled

Crashed Thread:        11

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 4 Illegal instruction: 4
Terminating Process:   exc handler [89250]

Application Specific Information:
BUG IN CLIENT OF LIBDISPATCH: Assertion failed: Block was expected to execute on queue [com.apple.main-thread]

Thread 0::  Dispatch queue: com.apple.main-thread
0   CoreGraphics                  	    0x7ff821f103e2 CG::DisplayList::colorResourceForColor(CGColor*) + 0
1   CoreGraphics                  	    0x7ff821f16b83 CG::DisplayListRecorder::getTextColorResourcesAndUpdateFlags(CG::DisplayList*, CGGState*, CGTextDrawingMode, CG::DisplayListResourceColor const*&, CG::DisplayListResourceColor const*&, unsigned int&) + 133
2   CoreGraphics                  	    0x7ff821f162d6 CG::DisplayListRecorder::DrawGlyphs(CGRenderingState*, CGGState*, CGAffineTransform const*, unsigned short const*, CGPoint const*, unsigned long) + 464
3   CoreGraphics                  	    0x7ff821ea2e0a CGContextDelegateDrawGlyphs + 376
4   CoreGraphics                  	    0x7ff822248691 draw_glyphs + 798
5   CoreGraphics                  	    0x7ff821ea2bdb CGContextShowGlyphsWithAdvances + 504
6   CoreText                      	    0x7ff81e31e7e4 EnumerateOverlappingGlyphs(CGContext*, TFont const&, unsigned short const*, long, int, void (CFRange, bool) block_pointer) + 173
7   CoreText                      	    0x7ff81e2a893d CTFontDrawGlyphsWithAdvancesInternal + 2706
8   UIFoundation                  	    0x7ff82055308e -[NSCoreTypesetter _NSFastDrawString:length:attributes:paragraphStyle:typesetterBehavior:lineBreakMode:rect:padding:graphicsContext:baselineRendering:usesFontLeading:usesScreenFont:scrollable:syncAlignment:mirrored:boundingRectPointer:baselineOffsetPointer:drawingContext:] + 2986
9   UIFoundation                  	    0x7ff8205544b1 -[NSCoreTypesetter _stringDrawingCoreTextEngineWithOriginalString:rect:padding:graphicsContext:forceClipping:attributes:stringDrawingOptions:drawingContext:stringDrawingInterface:] + 1361
10  UIFoundation                  	    0x7ff82048091a __NSStringDrawingEngine + 2373
11  UIFoundation                  	    0x7ff8204aed48 _NSStringDrawingCore + 173
12  AppKit                        	    0x7ff81f78087a _NSDrawTextCell + 3523
13  AppKit                        	    0x7ff81f81a203 -[NSTextFieldCell drawInteriorWithFrame:inView:] + 680
14  AppKit                        	    0x7ff81f819f4d __40-[NSTextFieldCell drawWithFrame:inView:]_block_invoke + 342
15  AppKit                        	    0x7ff81f616665 +[NSAppearance _performWithCurrentAppearance:usingBlock:] + 66
16  AppKit                        	    0x7ff81f819dc5 -[NSTextFieldCell drawWithFrame:inView:] + 194
17  AppKit                        	    0x7ff81f8cd339 -[NSTableView _drawContentsAtRow:column:withCellFrame:] + 251
18  AppKit                        	    0x7ff81f976120 -[NSOutlineView _drawContentsAtRow:column:withCellFrame:] + 71
19  AppKit                        	    0x7ff81f8cced4 -[NSTableView drawRow:clipRect:] + 1638
20  AppKit                        	    0x7ff81f8cc54d -[NSTableView drawRowIndexes:clipRect:] + 709
21  AppKit                        	    0x7ff81f974826 -[NSOutlineView drawRowIndexes:clipRect:] + 94
22  AppKit                        	    0x7ff81f8572cf -[NSTableView drawRect:] + 1670
23  AppKit                        	    0x7ff81f785fc3 _NSViewDrawRect + 121
24  AppKit                        	    0x7ff81fea033c -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1832
25  AppKit                        	    0x7ff81f7856c1 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 747
26  AppKit                        	    0x7ff81f785398 -[NSView(NSLayerKitGlue) drawLayer:inContext:] + 298
27  QuartzCore                    	    0x7ff823c8834e CABackingStoreUpdate_ + 613
[... 543 lines cut ...]
}
}

Model: MacBookPro16,1, BootROM 1715.40.15.0.0 (iBridge: 19.16.10549.0.0,0), 8 processors, 8-Core Intel Core i9, 2.4 GHz, 32 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Display: Color LCD, 3072 x 1920 Retina, Main, MirrorOff, Online
[... cut ...]
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
</details>

eric-wood, 2021-12-21:
Update: I upgraded to the latest nightly and was still experiencing the crash. I started messing with the code and commenting out some widgets to see if I could narrow down what was causing the issue. After getting nowhere for a bit I reverted back to the latest on the `main` branch and it seems to work now. I've restarted KiCad a few times and it seems to work consistently now.

So that's all to say I'm extremely confused as to why it works now. I'm on c59a983.

The only thing I really did that could have had an effect on this: my editor encouraged me to run `pip install -U autopep8`, which I did. That shouldn't have had any effect on this, though.

Sorry if this is not helpful!

Bouni, 2021-12-22:
@eric-wood As https://github.com/Bouni/kicad-jlcpcb-tools/commit/c59a9833174ca35e9084ca2e9676342cd72177a3 includes the potential fix for this issue https://github.com/Bouni/kicad-jlcpcb-tools/commit/525e8544eb22afb8aebc61d3f17bbb038c2b5d3a 
It seems that the fix works. Maybe @DanielO can verify that as well just to make sure.

DanielO, 2021-12-22:
Similarly it works for me but I already thought that was the version I had..

Very peculiar, but I guess you can close the issue :)

Bouni, 2021-12-22:
Perfect, thanks for testing 👍🏽

closed 2021-12-22 by Bouni

p1onk, 2021-12-28:
I'm still experiencing this / a similar issue on macOS. 

Abridged backtrace:
```
(lldb) bt
* thread #19, stop reason = EXC_BAD_INSTRUCTION (code=EXC_I386_INVOP, subcode=0x0)
  * frame #0: 0x00007fff201b025d libdispatch.dylib`_dispatch_assert_queue_fail + 99
    frame #1: 0x00007fff201b01f2 libdispatch.dylib`dispatch_assert_queue + 122
    frame #2: 0x00007fff28697d80 HIToolbox`islGetInputSourceListWithAdditions + 119
    frame #3: 0x00007fff28697ce1 HIToolbox`TSMGetInputSourceCountWithFilteredAdditions + 35
    frame #4: 0x00007fff28696b8e HIToolbox`TISCreateInputSourceList + 87
    frame #5: 0x00007fff286da67c HIToolbox`_TISCopyParentInputMethodForBundleID + 115
    frame #6: 0x00007fff288f708d HIToolbox`-[IMKInputSession dismissFunctionRowItemTextInputView] + 157
    frame #7: 0x00007fff288e62c2 HIToolbox`__71-[IMKInputSession imkxpc_dismissFunctionRowItemTextInputViewWithReply:]_block_invoke + 713
    frame #8: 0x00007fff2044fe52 CoreFoundation`__CFRUNLOOP_IS_CALLING_OUT_TO_A_BLOCK__ + 12
    frame #9: 0x00007fff2044fd03 CoreFoundation`__CFRunLoopDoBlocks + 440
    frame #10: 0x00007fff2044f06c CoreFoundation`__CFRunLoopRun + 2589
    frame #11: 0x00007fff2044df8c CoreFoundation`CFRunLoopRunSpecific + 563
    frame #12: 0x00007fff286dc096 HIToolbox`-[IMKInputSessionXPCInvocation invocationAwaitXPCReply] + 470
    frame #13: 0x00007fff286de19e HIToolbox`-[IMKInputSession deactivate] + 2976
    frame #14: 0x00007fff286dd5f1 HIToolbox`IMKInputSessionDeactivate + 36
    frame #15: 0x00007fff286dd5bd HIToolbox`DeactivateInputMethodInstance + 93
    frame #16: 0x00007fff286dd49a HIToolbox`utDeactivateAllSelectedIMInDocIterator + 82
    frame #17: 0x00007fff204147bf CoreFoundation`CFArrayApplyFunction + 62
    frame #18: 0x00007fff286b4e21 HIToolbox`utDeactivateAllSelectedIMInDoc + 132
    frame #19: 0x00007fff286b47f5 HIToolbox`MyDeactivateTSMDocument + 331
    frame #20: 0x00007fff286b4685 HIToolbox`DeactivateTSMDocument + 75
    frame #21: 0x00007fff22e6ff0e AppKit`-[NSTextInputContext deactivate] + 254
    frame #22: 0x00007fff22c5be6e AppKit`+[NSTextInputContext currentInputContext_withFirstResponderSync:] + 294
    frame #23: 0x00007fff22dfa3ff AppKit`-[NSTextView(NSSharing) setEditable:] + 214
    frame #24: 0x0000000100cad20f libwx_osx_cocoau-3.1.5.dylib`wxNSTextViewControl::WriteText(wxString const&) + 111
    frame #25: 0x0000000100c1e63e libwx_osx_cocoau-3.1.5.dylib`wxTextEntry::WriteText(wxString const&) + 78
    frame #26: 0x00000001774bdfe1 _core.cpython-38-darwin.so`meth_wxTextEntry_WriteText(_object*, _object*, _object*) + 161
    frame #27: 0x0000000101cc6d18 Python`cfunction_call_varargs + 120
    frame #28: 0x0000000101cc6785 Python`_PyObject_MakeTpCall + 373
    frame #29: 0x0000000101d94ed5 Python`call_function + 533
    frame #30: 0x0000000101d91c8d Python`_PyEval_EvalFrameDefault + 25677
    frame #31: 0x0000000101cc70d0 Python`function_code_fastcall + 128
```

This is both for the current `main` branch and also with abf2147.

Bouni, 2021-12-29:
@p1onk 
Can you please post you KiCAD Version info

p1onk, 2021-12-29:
Sure:

```
Application: KiCad Schematic Editor

Version: (6.0.0-0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.41.0

Platform: macOS Big Sur Version 11.6.1 (Build 20G224), 64 bit, Little endian, wxMac

Build Info:
	Date: Dec 24 2021 21:51:28
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.6.0
	Curl: 7.64.1
	ngspice: 35
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```