Bouni/kicad-jlcpcb-tools #92: Library "Package" search filter shouldn't be exact

riskable, 2021-12-22:
Searching for "SOT-23" in the Package field only shows parts that have that exact package name when it should show parts with "SOT-23-3", "SOT-23-4", "SOT-23-5", etc.

[image]

IMHO the search filter should be partial match by default with maybe the option to use quotes for an exact match.

riskable, 2021-12-22:
I just realized: This might have nothing to do with your code but might be a limitation of the JLCPCB API. If that's the case you can just close this since there's nothing you can do about it :shrug:

Bouni, 2021-12-23:
You can use `%` as wildcard operator to find packages that contain that string. For example `SOT-23%` will match `SOT-23` as well as `SOT-23-5`.
Click on the help button, there is a description!

[image]

closed 2021-12-24 by Bouni