Bouni/kicad-jlcpcb-tools #93: On make file gerber can rename .gbr to .gtl .gbl and all ?

Witawat, 2021-12-23:
On make file gerber can rename .gbr to .gtl .gbl and all ?

            pcbnew.F_Cu:      '[boardProjectName].GTL',
            pcbnew.B_Cu:      '[boardProjectName].GBL',
            pcbnew.F_SilkS:   '[boardProjectName].GTO',
            pcbnew.B_SilkS:   '[boardProjectName].GBO',
            pcbnew.F_Mask:    '[boardProjectName].GTS',
            pcbnew.B_Mask:    '[boardProjectName].GBS',
            pcbnew.Edge_Cuts: '[boardProjectName].GML',
            pcbnew.In1_Cu:    '[boardProjectName].G1',
            pcbnew.In2_Cu:    '[boardProjectName].G2',
            pcbnew.In3_Cu:    '[boardProjectName].G3',
            pcbnew.In4_Cu:    '[boardProjectName].G4',

thankyou

Bouni, 2021-12-23:
Why is that necessary?

arturo182, 2021-12-23:
This is possible with this checkbox in the Plot dialog, quite sure it's also possible using the Python API

[image]

I think some old PCB houses used the extensions to determine which layers are which, but most ones these days understand the KiCad layer names and don't really need this anymore. However having this as an option could be nice.

Witawat, 2021-12-24:
> Why is that necessary?

some software can't display layer *.gbr

Bouni, 2021-12-24:
I could easily add this by tossing in a `popt.SetUseGerberProtelExtensions(True)` in this section 
https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L85-L113 
But to make it selectable by the user that would require a checkbox somewhere and I don't want to clutter the main window with more options.
I think I create a new modal that comes up on clicking the generate Gerber button. There you can select various option and click an actual generate button to generate the files.
Or maybe its time for a settings menu? 🤔 
Not sure how to deal with this kind of things ...

Witawat, 2021-12-25:
[image]
add SetUseGerberProtelExtensions button on it

Bouni, 2021-12-27:
@Witawat That is exactly what I don't want! I do not want button on the main UI that just a very limited fraction of users will ever need. 
I'll create som sort of settigs menu where you can set such options soon.

bvernoux, 2021-12-27:
I'm totally agree the point is to generate gerber which work fine with JLCPCB and actual plugin do exactly that
I clearly do not see the point to add an other format which will bring nothing (except probably some problems with JLCPCB...)

Bouni, 2022-01-10:
If you need this you can add this line `popt.SetUseGerberProtelExtensions(True)` in `fabrication.py` somewhere in the `generate_gerber` method: https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L85-L113

Maybe I'll add this to a settings menu later.

closed 2022-01-10 by Bouni