Bouni/kicad-jlcpcb-tools #94: Plugin does not show up in menues

tronta, 2021-12-26:
I (hopefully) installed the plugin correctly on KI-Cad 6.0 by cloning it as described in
- /home/my-folder/.var/app/org.kicad.KiCad/data/kicad/6.0/scripting/pluginskicad-jlcpcb-tools/
- /home/my-folder/.local/share/kicad/6.0/plugins/kicad-jlcpcb-tools/

I tried flatpak and Fedora 35/Gnome  41.2 and little earlier versions, but don't see any additional icons/buttons which could be used.

Bouni, 2021-12-27:
Not sure if these folders are correct anymore.
Try to add a custom repo in PCM and install it from there:

[image]

tronta, 2021-12-27:
Thanks a lot. I was able to add the repo and install the plugin as you showed.
[image]

But unfortunately it still does not show up:
[image]

OnesAndZer0s, 2021-12-28:
for the sake of `copy-and-paste`-ing

`https://raw.githubusercontent.com/Bouni/bouni-kicad-repository/main/repository.json`

Bouni, 2021-12-28:
@tronta 
Please go to Einstellungen -> Einstellungen -> Action Plugins and check if you see a attention mark in the lower section

[image]

If so, click it to see what error it shows

tronta, 2021-12-28:
In action plugins it does not show up, instead I see a yellow triangle with exclamation mark:
```
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/my-home/.var/app/org.kicad.KiCad/data/kicad/6.0/scripting/plugins/kicad-jlcpcb-tools/__init__.py", line 4, in <module>
    from .plugin import JLCPCBPlugin
  File "/home/my-home/.var/app/org.kicad.KiCad/data/kicad/6.0/scripting/plugins/kicad-jlcpcb-tools/plugin.py", line 5, in <module>
    from .mainwindow import JLCBCBTools
  File "/home/my-home/.var/app/org.kicad.KiCad/data/kicad/6.0/scripting/plugins/kicad-jlcpcb-tools/mainwindow.py", line 16, in <module>
    from .fabrication import Fabrication
  File "/home/my-home/.var/app/org.kicad.KiCad/data/kicad/6.0/scripting/plugins/kicad-jlcpcb-tools/fabrication.py", line 8, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

**********************************
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/my-home/.var/app/org.kicad.KiCad/data/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/__init__.py", line 4, in <module>
    from .plugin import JLCPCBPlugin
  File "/home/my-home/.var/app/org.kicad.KiCad/data/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 5, in <module>
    from .mainwindow import JLCBCBTools
  File "/home/my-home/.var/app/org.kicad.KiCad/data/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 16, in <module>
    from .fabrication import Fabrication
  File "/home/my-home/.var/app/org.kicad.KiCad/data/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/fabrication.py", line 8, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
```

Bouni, 2021-12-28:
Seems again the packaging (flatpack / rpm) does not include `requests`.
I filed an [issue](https://gitlab.com/kicad/code/kicad/-/issues/9644) about the same problem a while ago but back then it was the MacOS package that missed the request package.

It seems that you installation uses your systems python, so you could simply try to run `pip install requests` (maybe prefix it with sudo) and try again.

tronta, 2021-12-28:
`pip install requests` does not seem to have an impact as it seems to be already installed
```
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: requests in /usr/lib/python3.10/site-packages (2.26.0)
Requirement already satisfied: charset-normalizer~=2.0.0 in /usr/lib/python3.10/site-packages (from requests) (2.0.4)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3.10/site-packages (from requests) (3.2)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/lib/python3.10/site-packages (from requests) (1.26.6)
```
So probably the flatpak version seems to have an issue. For Fedora, there I have not found the 6.0 version, but only the nightly which is 6.9 or so. Would it make sense to open your ticket again?

Bouni, 2021-12-28:
Seems that KiCAD is using python3.9 (`/usr/lib/python3.9/...`) and your default python is python 3.10 (`/usr/lib/python3.10/`)

Try this `python3.9 -m pip install requests`

tronta, 2021-12-28:
I tried  `python3.9 -m pip install requests`, but python 3.9 was not installed on my machine, therefore I suspect that the 3.9 version is the one inside the flatpak. What do you think? Not sure how to open a shell inside a flatpak.

Bouni, 2021-12-29:
Thats maybe true, I've never used flatpack befor so I can only guess as well ...
Do you know where I can find info about the fedora package / the flatpack package?
I think they are not officialy from the folks at KiCAD.

tronta, 2021-12-29:
At least it's presented directly on the download page: https://www.kicad.org/download/

Bouni, 2021-12-29:
You're right I'll file an issue again.

tronta, 2021-12-29:
Thanks a lot!

```
Application: KiCad

Version: 6.0.0, release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.79.1-DEV GnuTLS/3.7.2 (NSS/3.68.1) (OpenSSL/1.1.1l) zlib/1.2.11 libidn2/2.3.1 nghttp2/1.44.0

Platform: Linux 5.15.11-200.fc35.x86_64 x86_64, 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Dec 25 2021 12:30:46
	wxWidgets: 3.1.5 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.78.0
	OCC: 7.5.1
	Curl: 7.79.1-DEV
	ngspice: 35
	Compiler: GCC 11.2.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON
```

Bouni, 2021-12-29:
The issue in the KiCAD repo: https://gitlab.com/kicad/code/kicad/-/issues/10162

Bouni, 2021-12-30:
Unfortunately my issue was closed as invalid. Thy do not package python dependencies on linux, they come from the system.

Please try open the python shell in KiCad and enter these two lines:
```
import sys
print(sys.executable)
```

That should print the path of the python interpreter that is used by KiCAD.
My guess is that its `/bin/python3.9` but lets make sure.

If so, try this: `/bin/python3.9 -m pip install requests`

tronta, 2021-12-30:
Yes, I have seen this. Thanks a lot for your effort. I will try it later.

UlfWetzker, 2022-01-01:
On my system I also have this problem.
I am using KiCAD as a flatpak under Manjaro, but the sandbox is missing the requests module.
When I open a debug shell into the KiCAD flatpak and change to the /usr/lib/python3.9/site-packages/ directory, it contains different modules than my host system.

Using the argument '--filesystem=' faltpak can access directories from the host, but unfortunately /usr/lib is blacklisted here and it didn't work that way :(

So I installed the requests module directly insight the flatpak.
To do this you have to install pip first and then the module using the debug shell.
```
flatpak run --command=sh org.kicad.KiCad//beta
python -m ensurepip --upgrade
/var/data/python/bin/pip3 install requests
```

After that I could start KiCAD via flatpak and the jlcpcb plugin worked.

tronta, 2022-01-01:
Hi @UlfWetzker, you are my life saver. Thanks a lot.
Maybe one should add, that those are three commands which have to be entered one by one.
@Bouni, maybe it would make sense to add this to the Readme. What do you think?

Bouni, 2022-01-01:
@UlfWetzker @tronta Makes absolutely sense to add that to the Readme!
Will do it next week, as I don't have a PC at hand right now.
I personally never used flatpack and therefore I'm more than happy when people like @UlfWetzker supply a solution ☺️

UlfWetzker, 2022-01-01:
Actually it is a very hacky solution :/

It would be a bit better if the KiCAD flatpack would already bring pip3.
Then you would probably only have to run one command and not switch to the debug shell.
`flatpak run --command=pip3 org.kicad.KiCad install --user requests`

Bouni, 2022-01-02:
I'll reopen the issue on gitlab and see what they think about it.

UlfWetzker, 2022-01-02:
I read your gitlab issue and there it was proposed to install the module during the plugin installation process. For this the plugin should provide a requrements.txt.
I don't have any experience with KiCAD plugins, but is there a way to run 'pip install -r requirements.txt' there?
Alternatively, one would have to do this from insight the Python code by using the subprocess module  (which is somehow again a bit hacky).
But anyway it would be useful if pip3 was already part of the flatpak.

Bouni, 2022-01-02:
@UlfWetzker I did install dependencies from within the addon using submodule but that does not work on OSX.
After switching from pandas (which required that install step) to SQLite it was no longer necessary.

But if there's no pip installed in flatpack, this does not work anyway.

I need to figure out a way to do this properly.

Bouni, 2022-01-24:
@tronta @UlfWetzker 

I've added the 3 commands to the README as this is the easiest fix for now.
Maybe we come up with a propper solution in the future.

closed 2022-01-24 by Bouni

ZoAndr, 2022-06-23:
Good day!
Thanks a lot for amazing plugin!
I found some problem. Let me share with you.

It is installed well. But when I try to call icon, it yields error message.
The error message contains text:
----------------------------------------
Traceback (most recent call last):

File
"...KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools/plugin.py",
line 21, in Run
    dialog = JLCPCBTools(None)
	
File
"...KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools/mainwindow.py"
line 569, in __init__
   self.init_store()
   
File
"...KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools/mainwindow.py"
line 585, in init_store
   self.populate_foootprint_list()
   
File
"...KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools/mainwindow.py"
0: wx.dataview.DataViewIconText(

TypeError: DataVIewIconText(): arguments did not match any overloaded call:
   overload 1: argument 2 has unexpected type 'Bitmap'
   overload 2: argument 1 has unexpected type 'str'
-------------------------------------------------------
I am using KiCad 6.0.4, release build, Windows 10
Thanks a lot!!!

auipga, 2022-12-12:
`python -m pip install requests` is also needed when installed from the custom repository but this step is only mentioned for the Flatpack install method.

```sh
$ python -V
Python 3.10.8

$ python3 -V
Python 3.10.8

$ python3.10 -V
Python 3.10.8

```

kiwi85, 2023-01-01:
hello all regarding the plugin error I found a workaround inside the pluginfolder in file fabrication.py:
delete the following lines:
`from pcbnew import (
    DRILL_MARKS_NO_DRILL_SHAPE,
    EXCELLON_WRITER,
    PCB_PLOT_PARAMS,
    PLOT_CONTROLLER,
    PLOT_FORMAT_GERBER,
    VECTOR2I,
    ZONE_FILLER,
    B_Cu,
    B_Mask,
    B_Paste,
    B_SilkS,
    Cmts_User,
    Edge_Cuts,
    F_Cu,
    F_Mask,
    F_Paste,
    F_SilkS,
    GetBoard,
    GetBuildVersion,
    In1_Cu,
    In2_Cu,
    In3_Cu,
    In4_Cu,
    Refresh,
    ToMM,
    wxPoint,
)`
and add:
`from pcbnew import*`