Bouni/kicad-jlcpcb-tools #95: Parts not being put into `part_info` database

OnesAndZer0s, 2021-12-27:
I'm running linux mint, downloaded plugin, ran plugin, no dice.

The `parts` database has entries, however. But, the `meta`, `rotation`, and `part_info` databases do not.

https://user-images.githubusercontent.com/37850372/147495010-612060b2-23f7-44bd-a0d4-7fdc24ed5b82.mov
I've added some debug entries in the logger. Currently still digging through the source to see why this is happening

Bouni, 2021-12-28:
The rotations db is empty by default, by clicking Manage rotations -> Update the rotations CSV is downloaded from Github and imported into the database.

The `part_info` table in the `project.db` is normally populated once you open the plugin, it reads the `.kicad_pcb` file via the API and puts all parts into it.

You should see output like this in the bottom logger:

[image]

Bouni, 2021-12-29:
@OnesAndZer0s Any news on this?

OnesAndZer0s, 2021-12-30:
It works fine, I'm just a mega idiot. I was under the impression that you searched a part from the database and then put it on the sketch, not the other way around. Probably should specify that in the "how to use"

Bouni, 2021-12-30:
No problem, happy that it works now!

> Probably should specify that in the "how to use"

Yes, documentation need improvement 😅

closed 2021-12-30 by Bouni