Bouni/kicad-jlcpcb-tools #96: freeze

radontec, 2021-12-28:
I´ve problems with a newer version of the plugin and kicad.

[image]

Kicad freeze every time I try to start the plugin.

Bouni, 2021-12-28:
Please close KiCAD, open the Terminal first, then open the plugin. Check if there's any output in the stdout/stderr window

radontec, 2021-12-28:
[image]

the console freezes too

Bouni, 2021-12-28:
Please post the output of Help -> About KiCad -> Copy Version Info

[image]

radontec, 2021-12-28:
this is now almost one of the latest versions of Kicad. However, I have a slightly older version on another computer with which this used to work well. Only with one of the newer versions of the plugin there is this problem. 

Application: KiCad Leiterplatteneditor (64-bit)

Version: (6.99.0-39-gdb06693d22), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 22000), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 24 2021 09:01:15
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

Bouni, 2021-12-28:
This is strange, I've installed the latest nightly just minutes ago on a Win10 machine as well, installed the plugin via PCM and opened it on a random project ant everything works just fine for me 🤔

Bouni, 2021-12-28:
```
Application: KiCad (64-bit)

Version: (6.99.0-122-g9ba733f23f), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 27 2021 20:59:59
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

The only difference I can spot is that you have a slightly newer version of windows.

radontec, 2021-12-28:
I tryed two different computer (one with Win11 and one with Win10) 
I think the problem is our pcb or the components itself

I´ll try an older version of the plugin

radontec, 2021-12-28:
I used the version from 08 of September and with this old version of the plugin its working

Bouni, 2021-12-28:
Its hard to figure out whats the case if there are not errors and its not reproducable on my side.

I'll try to compare these versions and see if I find the difference but that will take some time ...

radontec, 2021-12-28:
file via wetransfer:
https://wetransfer.com/downloads/cd6a0111225c6df70e5654f7cc4992fe20211224145108/cf9b2a

would be interesting if you have the same problem

shadowlab, 2021-12-29:
To chime in here as well, I'm having the same issue while running under Windows 11. It does however, seem to be project specific, as I was able to load a smaller board design without issue. My guess is complexity? The project that loaded OK has maybe 1/4 the overall components than the larger board has. 

Application: KiCad (64-bit)

Version: (6.0.0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 22000), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 24 2021 19:17:02
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

Bouni, 2021-12-29:
@radontec I've just doanloaded your project, opened it, opened the plugin and it works :expressionless:

[image]

I'll try to update my windows, maybe I get the error to reproduce that way

Bouni, 2021-12-29:
I'v updated my windows machine to the latest available version but after that I'm still on `Win 10 Pro 21H2 Build 19044.1415`
I guess my Hardware is to old to get any newer versions of windows

Bouni, 2021-12-29:
@shadowlab What is a complex project in terms of footprint count in your eyes?
Can you send me a project that triggers this problem so that I can have a look at it?

radontec, 2021-12-29:
tried it with a third computer, same problem and on this machine an older Windows version is running
@Bouni if you like I can give you anydesk access

Application: KiCad PCB Editor (64-bit)

Version: (6.0.0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 24 2021 19:17:02
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

Bouni, 2021-12-29:
@radontec  Let me finish my lunch first. After that I'll take a look via anydesk (send me the login data to bouni @ owee . de)

radontec, 2021-12-29:
no hurry, just connect when you are ready

Bouni, 2021-12-29:
OK after a debugging session with @radontec (thanks for the help by the way) I figured out that if I comment out this line https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/store.py#L31 wait for the database to be initially downloaded and comment the line back in again everything works.

I guess its some sort of race condition but I'm not yet sure where to look for it. Still we're getting closer!

Bouni, 2021-12-29:
Ok, I'm finally able to reproduce the issue. Let's see if I can figure out whats the root cause

radontec, 2021-12-29:
I just noticed one more thing, the generated Gerber file does not work with jlcpcb. i am checking now if this is coincidence and if it works with the old version.

radontec, 2021-12-29:
hmm the Rotation tool looks strange and this is a new function, compared to the old plugin version
[image]

Bouni, 2021-12-29:
@radontec Yes you can manage in the UI how much additional rotation a part should be applied with to fit JLCs part rotations, they are different from KiCADs.
By the way, please don't spam issues with arbitrary questions, open a seperate issue if you have something thats not related to the issue itself.

shadowlab, 2021-12-30:
> @shadowlab What is a complex project in terms of footprint count in your eyes? Can you send me a project that triggers this problem so that I can have a look at it?

It seems with the newest release, the UI opens and doesn't freeze any longer. If it happens again, I'll open up a new ticket. 

Thanks for all your hard work!

Bouni, 2021-12-30:
Thanks for reporting back @shadowlab

radontec, 2021-12-30:
perfect, the fix does what it should, it has now worked at first go. Thank you very much!

bask185, 2023-08-23:
I have a question. I am still having this issue 2 years later. The Plugin freezes up Kicad for a few minutes. I am not sure if I should open a new issue for this.

I have learned that the freeze occurs if atleast one component has an LCSC number. The freeze starts again as soon if you modify something using this tool.

The freeze occurs on only one of my 2 PC. The PC on which it happens, is my work PC and the slow network may or may not be of relevance.

[image]

Application: KiCad x64 on x64

Version: 7.0.7, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Aug 14 2023 02:42:39
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.7.1
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

Bouni, 2023-08-24:
Would you please update the plugin to the most recent version to rule out that this issue is already fixed by a newer version?

There are already a couple of newer versions available:

[image]

bask185, 2023-08-24:
I updated and there is a small change. Kicad itself does not freeze but it still takes forever. I do can tell how long exactly
[image]

Like I suspected, it is most likely a slow internet problem. The debug screen shows me a path which I don't use. Some IT clown decided that my default document folders should be in the network instead of having it local.

I can see that the path leads to our network storage and not my local C disc.
[image]

I don't know in detail how things work like you do. I don't know where I can change this behaviour and configure a different path

bask185, 2023-08-24:
This new version seems not to work at all anymore. I don't have any part number anymore. And I have not changed my routine
[image]

Bouni, 2024-08-27:
Re-open if this issue still exists