Bouni/kicad-jlcpcb-tools #97: grouped components ?

AiGreek, 2021-12-31:
yeah, it's me, again :D 
I'm making a HUGE pcb with a lot of LED, header, same resistors, capacitors, etc ... and i was wonrdering if it's possible to group components by value to add LCSC reference once without having to scroll for 10 years each time..

This what we have : 
[image]

This is what we COULD have : 
[image]

Not sure if it's easy to do, or even if it's useful for anyone except me but meh... why not ?!

Bouni, 2021-12-31:
The `Select alike` button is your friend 😉

arturo182, 2021-12-31:
Maybe "Select similar" would've been more intuitive 🤔

Bouni, 2021-12-31:
Naming was suggested by a native speaker (at least I guess he is) 🤷

AiGreek, 2022-01-01:
> The `Select alike` button is your friend 😉

I think I misunderstood myself. It is not a question of component selections, but of display of components in the window. I know this button exists (and indeed, the English translation is weird) ^^

Bouni, 2022-01-01:
I have to look into that but I guess that's rather hard to implement. WX lists are a pain in the rear if you use them normally, so having this type of ui sugar is extremely difficult.

Bouni, 2022-01-10:
I think I'll not implement ths for now as the `select alike` button is good enough in my opinion. I think I'll go for @arturo182's suggestion and change the button label from `select alike` to `select similar`

closed 2022-01-10 by Bouni