Bouni/kicad-jlcpcb-tools #98: Cannot open part datasheet from Flatpak

tronta, 2022-01-01:
**Describe the bug**
It is not possible to open the data sheet from the part details.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Open JLCPCB Tools
2. Search and select a part
3. Click on Parts details
4. JLCPCB Part Details open
5. Click on Open Datasheet --> not possible
6. Close JLCPCB Part Details --> not possible
7. Dialog closes when closing JLCPCB Tools dialog

**Expected behavior**
I assume, browser opens the expected page

# KiCad Version

```
Application: KiCad Leiterplatteneditor

Version: 6.0.0, release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.79.1-DEV GnuTLS/3.7.2 (NSS/3.68.1) (OpenSSL/1.1.1l) zlib/1.2.11 libidn2/2.3.1 nghttp2/1.44.0

Platform: Linux 5.15.12-200.fc35.x86_64 x86_64, 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Dec 25 2021 12:30:46
	wxWidgets: 3.1.5 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.78.0
	OCC: 7.5.1
	Curl: 7.79.1-DEV
	ngspice: 35
	Compiler: GCC 11.2.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON
```

tronta, 2022-01-01:
I assumed that some rights are missing in the manifest, so I added all (?!) rights to it via Flatseal, but it didn't solve the issue.

Bouni, 2022-01-02:
Oh boy, flatpack is becoming a nightmare I guess 😵
I'll look into it the coming week