Bouni/kicad-jlcpcb-tools #100: can't compress Gerber or SMT Assembly files

AiGreek, 2022-01-09:
Hiiiiiiii ^^'

Don't know why or when or how, but since the last update of Kicad Nightly (?) the plugin seems unable to zip or create the files for the SMT assemblies (BOM/POS). The logs do not seem to indicate any error whatsoever.

[image]

```
Application: KiCad PCB Editor (64-bit)

Version: (6.99.0-294-g04c46a8fd2), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan  7 2022 21:00:04
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-01-10:
I have not yet checked how the plugin is working with 6.99 so I concider it 6.0 only at the moment!

Bouni, 2022-01-12:
I looked into this because I decided that (if possible) I want to support 6.0  as well as 6.99 becaus ein 6.99 a ton of cool feature land in the coming days an months.

I found that the API changed slightly in 6.99 and I check which version is running and use the right type (wxPoint vs. VECTOR2I) to avoid the problem.

commit 00c06c5d (2022-01-12): Fix #100, 6.99 API change
closed 2022-01-12 by Bouni