Bouni/kicad-jlcpcb-tools #101: failed to download the JLCPCB database CSV - Jan 10th 2022

lyso1, 2022-01-11:
**Describe the bug**
I installed the plugin from the plugin manager in Kicad v6 but I can't select parts at all and the plugin just displays the "failed to download the JLCPCB database CSV, error code 404" pop up error. I know that jlc did have issues with their part picker recently so I'm hoping it's just that.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the kicad v6 plugin manager, add the URL to the repository
2. Click on 'install'
3. open PCB design to use the plugin on
4. click the plugin and See pop-up error

**Expected behavior**
to be able to select and search for parts within jlcpcb/lcsc and assign them to each footprint

# KiCad Version

```Application: KiCad PCB Editor (64-bit)

Version: (6.0.0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 24 2021 19:17:02
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

lyso1, 2022-01-11:
no warning in action plugins either btw
[image]

Bouni, 2022-01-11:
This is a JLCPCB issue with their webservers hosting the CSV database. Should be working again.

closed 2022-01-11 by Bouni