Bouni/kicad-jlcpcb-tools #102: Component search not working

lpdx, 2022-01-11:
**Describe the bug**
When I try to search for a component nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Open J@LC plugin
2. Click in "Update library"  (it shows me that a CSV file was downloaded).
3. Click in some component
3. Click in "Select part"
4. JLCPC Library window opens to search
5. I put some values to search: keyword=10k and Category=Resistor
6. Click "Search"
7. Nothing happens. 

**Expected behavior**
A list of components to choose.

# KiCad Version
```
Application: KiCad PCB Editor (64-bit)
Version: (6.0.0), release build
Libraries:
    wxWidgets 3.1.5
    libcurl/7.78.0-DEV Schannel zlib/1.2.11
Platform: Windows 10 (build 22000), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Date: Dec 24 2021 19:17:02
    wxWidgets: 3.1.5 (wchar_t,wx containers)
    Boost: 1.76.0
    OCC: 7.5.0
    Curl: 7.78.0-DEV
    ngspice: 35
    Compiler: Visual C++ 1929 without C++ ABI
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

[image]

Bouni, 2022-01-11:
Have you tried just `10k` as a keyword without a category?

lpdx, 2022-01-11:
No I din't. 
I see now that without a category it works.

Thank you!

(the pre filled values induces the user to the wrong example)

closed 2022-01-11 by lpdx