Bouni/kicad-jlcpcb-tools #106: Components position units and board units seem different.

rsre, 2022-01-17:
**Describe the bug**

I created the production files but JLC PCB fails to properly place the components.
[image]

It seems like the units for the .pos file and board gerbers don't match. Error accumulates the further the component is from the origin. I tried to edit the units and to generate the .pos and gerber files by hand but I did not succeed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to plugin menu
2. Click on 'Generate Fabrication Files'
3. Upload them to JLC PCB
4. See error

**Expected behavior**
The components are shown placed in their correct positions on JLC PCB preview.

# KiCad Version

```
Application: KiCad PCB Editor

Version: (6.0.1-0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.77.0 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.42.0

Platform: macOS Version 12.1 (Build 21C52), 64 bit, Little endian, wxMac

Build Info:
	Date: Jan 14 2022 14:47:36
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.6.0
	Curl: 7.64.1
	ngspice: 35
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-01-20:
I just did a quick test with a very simple project using KiCAD 6.0.1

```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.1), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan 15 2022 13:36:27
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

The positions are all correct, there are several with inventory shortage and therefore are not displayed but the positions seem to fit.

[image]

So I guess its either an issue with the board or with the MacOS Version!?

Can you somehow send me your files so that I can check them on my side?

Bouni, 2022-01-20:
Oh, and whats the result if you export the CPL file via the KiCAD UI?

[image]

Maybe its some sort of issue with the units:

[image]

Please test if you can get the same result by either using `mm` or `inch` (you need to manually modify the resulting CSV in order to make it compatible with JLC, I can do that for you if you don't know how)

rsre, 2022-01-20:
I did some more testing and apparently the problem is due to an USB connector hanging over the board's edge. This is necessary because the board fits inside an enclosure and the connector has to poke through.

I don't know if it's just that JLC PCB doesn't allow components to extend over the board's edge or that the files are not generated right.

When I tried without assembling the USB connector it worked fine.

Bouni, 2022-01-20:
This is very starnge! I would totally understand if a single part is displaced due to overhang but why does it affect all the parts!?
Anyway, glad that you've figured it out and it works again 😄

closed 2022-01-20 by Bouni