Bouni/kicad-jlcpcb-tools #110: Problem with HiRes monitor

glottis512, 2022-01-21:
**Describe the bug**
DPI scaling is not optimal on hi dpi monitor

[image]

**To Reproduce**
Run plugin on hi dpi monitor with scaling

# KiCad Version
Application: KiCad editor DPS (64-bit)

Version: (6.0.1), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan 15 2022 13:36:27
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

Bouni, 2022-01-24:
After reading a bit about High DPI support in wxpython it seems that this in particular is a pain in the rear to get working correctly.
I'll see if I can get it to work but will take a while, so please be patient 🙏🏽

Bouni, 2022-01-24:
@glottis512 I found a solution and will merge the PR soon after I've tested a bit more.
If you like you can test it as well and give me feedback if it works for you as expected.

glottis512, 2022-01-24:
really? so fast? :) awesome. yes, i can test it as well. 

i tried clone this branche into proper directory but without success. This is the error. But this could be my fault.

[image]

Bouni, 2022-01-25:
💩 So 6.99 seems to package a different wx version than 6.0 😭

Bouni, 2022-01-25:
Ok, I figured out a way that should work for both, 6.0 and 6.99 🙏🏽 

Please be so kind and pull and test again 😃

glottis512, 2022-01-25:
now is it perfect, sir :) Thank you!

Michal

[image]

Bouni, 2022-01-25:
Ok, then I'll merge this and release a new version

closed 2022-01-25 by Bouni

DevynCJohnson, 2022-07-08:
@Bouni , this may or may not be of any help, but on my system for any widgets that are not scaling properly, I set the below property

`HighResWxSize(parent.window, wx.Size(-1, -1)),`

I changed the enclosed numbers from whatever they were to `-1`.