Bouni/kicad-jlcpcb-tools #111: Plugin threw an exception on MacOS.

eTLAsw, 2022-01-21:
**Describe the bug**
When I press the JCL button in the PCB editor the plugin throws an exception.
KiCad stays running though.

**To Reproduce**
Steps to reproduce the behavior:
1. Added the repo and installed the plugin through the PCM following the instructions in the README.
2. Restarted KiCad.
3. Pressed the button.
4. Got  this exception:
```
Traceback (most recent call last):

File "/Users/martin/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 20, in Run
dialog = JLCBCBTools(None)

File "/Users/martin/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 410, in __init__
self.init_store()

File "/Users/martin/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 424, in init_store
self.populate_footprint_list()

File "/Users/martin/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 489, in populate_footprint_list
details = self.library.get_part_details(numbers)

File "/Users/martin/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/library.py", line 244, in get_part_details
return cur.execute(

sqlite3.OperationalError: no such table: parts
```

**Expected behavior**
To be able to select parts according to the nice little instruction video. :) 

# KiCad Version

```
Application: KiCad

Version: (6.0.1-0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.77.0 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.42.0

Platform: macOS Version 12.1 (Build 21C52), 64 bit, Little endian, wxMac

Build Info:
	Date: Jan 14 2022 14:47:36
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.6.0
	Curl: 7.64.1
	ngspice: 35
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

The computer is an iMac 27" with an i9 CPU.

Let me know if you need additional help/information.

dgoodlad, 2022-01-22:
I just hit exactly the same issue; `jlcpcb/parts.db` is a 0-length file. Seems none of the tables are getting created on first start...

dgoodlad, 2022-01-22:
I somewhat reversed the change from f0ad3909cc096ac811e7451b2024a967416cb935 by calling `self.check_library` in `setup`; this at least triggered the download before the crash. It did still 💥 once, then popped up a message saying it had updated the library 15s later, and it works fine now. I haven't got the python/kicad scripting experience to dig deeper, but this should at least point someone in the right direction 😄

eTLAsw, 2022-01-22:
Ah. Don't think I want to go into messing around either, but it sounds like a not too hard problem to solve then. Hopefully it gets fixed soon. :)

Bouni, 2022-01-24:
I see where the problem is. When a user starts the plugin for the first time, there is no parts database to get data from for populating the footprint list. We need to download the library befor we can do that.
I work on a fix, maybe I can release it today ...

eTLAsw, 2022-01-24:
Excellent!

I look forward to trying it out. The video looks really sweet.

(And no hurry for my sake, I'm not even close to being able to order a PCB yet.)

closed 2022-01-24 by Bouni