Bouni/kicad-jlcpcb-tools #117: Download JLCPCB/LCSC symbol & footprint library?

glowingkitty, 2022-01-27:
Thanks, awesome work with this plugin! It's already super helpful and saves a lot of time!:)

But did you consider also downloading the symbols and footprints from LCSC via the plugin directly into KiCad? So one can use them just like regular parts and also use everything offline? Also makes it easier to directly just use parts which are on stock (and ideally Basic parts). Download size should still be reasonable for the whole library (maybe 2-3GB) If symbols and footprints only, without 3D model (or 3D model optional).
https://github.com/TousstNicolas/JLC2KiCad_lib might be a good starting point for integrating that feature.

Or is the problem the API rate limit of LCSC/JLCPCB for getting those data?

Bouni, 2022-01-28:
I thought about this in the past but there are several hurdles to overcome. 
First, there is no official public API to do that. What @TousstNicolas does with his JLC2KiCad_lib is using reverse engeneered API calls. Thats not a bad thing per se but I'm not sure If JLC/LCSC change it at some point which renders this functionality unusable.
So maintaining it can be hard. 
Also I don't want to "steal" @TousstNicolas idea/plugin and integrate it into mine 😄 
Maybe I can check if his plugin is installed and call it from mine 🤔 

The next "problem" is, once you download the symbol / footprint / 3d model you add it to a new or existing lib which requires a refresh I guess. Maybe possible for PCBnew aka. PCBeditor but at this point there's no way of interacting with eeschema / Schematic Editor via Python (But thats on the roadmap for v7 I read somewhere).

I'll see if @TousstNicolas  is interested to integrate his plugin into KiCAD and add a GUI, that way it would be much better to use and I can probably call it from my plugin.

glowingkitty, 2022-01-28:
„Stealing“?:D isn’t the whole point of open source software to build on top of it / improve it and this way saving a lot of time and creating even better open source software?:D so at least that I wouldn’t worry too much about. But yeah asking what @TousstNicolas thinks about this idea sounds great :) 
And fair enough… building on top of an reverse engineered API sounds indeed tricky^^ similar to building on top of a web scraper, which could crash any time the website makes changes to the way they show data on their website :D

TousstNicolas, 2022-01-30:
Integrating my script into a KiCad plugin with a gui and some extra functionality is something I've been thinking about for a while, and I've already tested some prototype interfaces. The main problem, as @Bouni raises, is the lack of interaction between a python plugin and EEschema.

I have never used your plugin, so I may not be in the best position to propose integration solutions, but I am totally open to proposals. 
From what I've seen, adding a button calling a function in my script to generate a list of selected footprints from your search tool might be feasible.

Bouni, 2022-06-08:
I've decided not to implement this for now, there are perfectly fine tools around that solve this already.

closed 2022-06-08 by Bouni