Bouni/kicad-jlcpcb-tools #118: Error: "Exception on python action plugin code"

LioBou, 2022-01-28:
**Describe the bug**
After plugin installation, first try by clicking the new icon: Error window "Exception on python action plugin code"

Text inside the error window:

**File "/home/myloginname/.local/share/kicad/6.0/3rparty/plugins/~com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 20, in Run Dialog = JLCPCBTools(None)
File "/home/myloginname/.local/share/kicad/6.0/3rparty/plugins/~com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 47, in __init__self.SetSize(self.Window.FromDIP(wx.Size(1300, 800)))
AttributeError:'JLCPCBTools' object has no attribute 'FromDIP'**

**To Reproduce**
Steps to reproduce the behavior:
1. Installed plugin from repo as described in ReadMe
2. Running KiCAD 6.0
3. Python 3.8.10 installed
4. Running on Linux Mint 20.3 Una (base: Ubuntu 20.04 focal)

**Expected behavior**
Start the plugin, but error window, nothing else.

# KiCad Version
```
Application: KiCad PCB Editor

Version: 6.0.1-79c1e3a40b~116~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.4.0-96-generic x86_64, 64 bit, Little endian, wxGTK, mate, x11

Build Info:
	Date: Jan 15 2022 20:26:22
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.81.0
	ngspice: 31
	Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

LioBou, 2022-01-28:
I wish I can use your plugin soon, looks great!

Bouni, 2022-01-31:
This is strange, the error message states `__init__self.SetSize(self.Window.FromDIP(wx.Size(1300, 800)))` 
`Window` is uppercase but in the code is clearly lowercase 🤔

Bouni, 2022-01-31:
Can you verify by opening this file `/home/myloginname/.local/share/kicad/6.0/3rparty/plugins/~com_github_bouni_kicad-jlcpcb-tools/mainwindow.py` in a text editor if `Window` is lower or upper case?

LioBou, 2022-01-31:
yes, "windows" is lower case in the code.
(I had to wrote the full error message myself as the error window do not allow selection and copy/paste, so this is me that wrongly write in upper case).

Bouni, 2022-01-31:
OK iguess thats a `wx` issue then.
On my Windows machine I have `wxWidgets: 3.1.5 (wchar_t,wx containers)` where you have `wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24`

I need to figure out if that wx version is coming with KiCAD or if thats installed in your systems python. You maybe have to upgrade that but that might break other things on your system :worried:

benaveld, 2022-01-31:
I got the same error on Manjaro kernal 5.15.16-1.
I tried upgraded wxWidgets to version 3.1.5. By doing that KiCad gives an error that the wrong version of wxWidgets is used.

BTW: wxWidgets: 3.0.5 is included as a dependency to KiCad.

Bouni, 2022-01-31:
@benaveld thanks for confirming this. 

Then it must be some kind of Windows / Linux differene that I don't see at the moment ...
I'll look into it as soon as possible.

freaf87, 2022-02-01:
Hi, 
I also got the same error in Ubuntu 20.04.3 with Kicad
Application: KiCad Version: 6.0.1-79c1e3a40b~116~ubuntu20.04.1, release build

[image]

Bouni, 2022-02-02:
Seems the FromDIP method was introduced in wxWidgets 3.1.0

https://docs.wxwidgets.org/trunk/classwx_window.html#aa934a31552633c076a87bca1100b0dc5

Bouni, 2022-02-02:
@LioBou @freaf87 @benaveld 
It would be awesome if one of you guys coud test the PR befroe I merge it into main.

LioBou, 2022-02-03:
Hi, thanks Bouni. I just tested (I basically replaced old files by the one on this branch, I'm not good yet with git), and I got a new error:
[image]

Bouni, 2022-02-03:
OK, GetDPIScaleFactor was intoduced even later, in 3.1.4: https://docs.wxwidgets.org/trunk/classwx_window.html#a22e9ffc6ad8aba15eecf5537f3b7791f
I'll add a workaround for that as well.

Bouni, 2022-02-03:
@LioBou  Please test again

LioBou, 2022-02-03:
Great, no error this time ! Seems to work ! I will try it more deeply.

Bouni, 2022-02-03:
Nice, thank you for testing! 
I'll wait till tomorrow and then merge this.

LioBou, 2022-02-04:
I was able to select parts and generate assembly and gerber files.
Perfect, many thanks ! I'll be able to test assembly by JLCPCB soon.

LioBou, 2022-02-04:
@Bouni, I'd like to sponsor you and participate to your "23" goal, but when entering the amount, the "select" button is not active... Any idea what's wrong?

Bouni, 2022-02-04:
@LioBou Highly appreaciate that you want to support me :star_struck:
I have to look into that issue, give me a few minutes

closed 2022-02-04 by Bouni

Bouni, 2022-02-04:
@LioBou I can not see anything wrong ...
If I log out of my account in order to see the actual sponsorship page I can click the select button:

[image]

Is that the select button that you cannot click?

Maybe you must add some sort of payment method to your github account, like PayPal for example!?

LioBou, 2022-02-04:
@Bouni Yes, it is the Select button which is not active. I added payement info, still the same... :(  I have a free GitHub account, might be linked to this status ? That would be weird.

Bouni, 2022-02-04:
I don't think so. I asked a college if he could sponsor me and for him it works. Both of us have a free account, he has no payment info configured.
What country are you from? Maybe that's the reason!?

LioBou, 2022-02-05:
I'm from France. I thought that the dollars unit would maybe block since I'm in euros country, but most likely not that.
The thing is that the Select button is active 0,5s when I switch from monthly to one time and vice versa, and then go back grey and inactive.
I'm using Chrome, i'll try with Firefox.

LioBou, 2022-02-05:
I found the root cause... which was... Addblockplus !!
Simply deactivated for GitHub to make it work.

Bouni, 2022-02-05:
🥳 glad you figured it out

Bouni, 2022-02-07:
@LioBou Apparently theres no way to say Thank you through GitHub Sponsors, so I'll do it here!

Thank you @LioBou  for your support 🤩

PR #124 (2022-02-04): Attempt to fix wxWidgets < 3.1 issue with HighDPI capabilities
This is a hacky attempt to fix #118.

It seems linux versions of KiCAD use wxWidgets 3.0.4 while Windows uses 3.1.5.
In order to support HighDPI Displays I introduced the use of the FromDIP method which, unfortunately is supported from wxWidgets 3.1.0 onwards.

This means that at the moment there's no High DPI support for linux after this PR has been merged but at least the plugin is usable again in linux.