Bouni/kicad-jlcpcb-tools #119: Popup resetting values and select similar

atlaste, 2022-01-29:
Most of the times I want the same settings (like 'basic part') for the next component. I also regularly forget to click 'all similar' before clicking the 'select component'.

Is it possible to retain the settings from the previous search and have a button for 'select similar' in the popup? Or even better: fill in the value and footprint separated by a space in the search box top-left automatically (perhaps even searching)? If it gives search results in the basic list, I don't even want the extended list in most of the times; I imagine this to be the general case.

Bouni, 2022-01-31:
I thought quite a few times how I ideally implement a settings or preferences window but didn't come up with something good yet.
I'll think a bit more how to do this and then implement a solution that takes your suggestin into account as well.

Bouni, 2022-02-08:
1. I have to think about how to implement this
2. I'm not sure if I can interact with the GUI in that way but if its possible I try to figure out a way to do it. Sound like a usefiul feature.

atlaste, 2022-02-08:
I'm just going on while I use your plugin :-) 

Another very useful feature that is probably easy to add is to cherry pick similar in the first dialog. Let me explain: Normally, you do 'select alike' and then 'select component'. However, when you modify the design and add/remove components, you often get new, similar components that should get the same LCSC code. Think f.ex. about decoupling caps.

So, what I would like is a button "Auto alike". It just grabs the first LCSC code that matches the same footprint/value, and fills it in (or in case of ambiguity don't fill it in).

Bouni, 2022-04-01:
- PR #152 saves the checkboxes as requested
 - Highlighting parts on the board is not possible to my knowledge (There has to be a way as this is what happens when you click a part in the schematic but maybe that function is not exposed in the Python API )

closed 2022-04-01 by Bouni