Bouni/kicad-jlcpcb-tools #120: Show basic / extended and price in table

atlaste, 2022-01-29:
While optimizing my board I usually want to know the price / piece and the fact that something is a basic / extended part. Can we show that in the table next to the part number, preferably sortable?

Bouni, 2022-01-31:
You already have the basic/extended info in the footprint list:

[image]

Sowhing the price is not that easy! there are no fixed price steps and even worse, parts can have a minimum quantity.
That means if you need lets say 5 of a certain part for your 5 boards, the charge you 20 because thats the minimum quantity.

This info is not part of their parts CSV database and I woul need to query it from an unofficial API which I want to prevent because I don't want users to hammer their API 😬

atlaste, 2022-02-08:
If I download the CSV directly from jlcpcb, it has a 'price' column with things like this:

20-180:0.004724409,200-580:0.003905512,600-1980:0.003307087,2000-9980:0.002818898,10000-19980:0.002724409,20000-:0.002692913

I suppose it sais '20-180 pieces go for 0.004724409 each, 200-580 for 0.003905512 each, etc' These upper bounds don't make sense. The 20 is probably the lower bound. But just putting in the # boards someone will order should give all the information to calculate the price per item. Extended parts without through holes have a 3,- charge. Through holes depends on the number of holes.

I'm not sure what I want here to be honest. I really don't care about the very cheap components like resistors and caps, like most people I think. It's mainly for expensive ones. For example, there are often multiple IC's I can use, and day prices change so rapidly that it depends on the day what component I want. For CAN i have a solder jumper in place that I can use or not to figure out which IC is best... Ideally I want rules like that, but that's definitely outside the scope.

Bouni, 2022-02-08:
The main problem is that you cannot simply say I've got 5 pcs. of x on my board and the price for 1-20 ist lets say 0.05USD so thats 6 x 0.05 = 0.3USD! JLC has a minimum quantity for some if not the most parts, which in this case could be, lets say 20. so the charge you 20 x 0.05 = 1USD even if they only populate 6 of them for you.
In order to get that number (20) I've to do a API request every time which I want to avoid because its not an official API.

atlaste, 2022-02-08:
> The main problem is that you cannot simply say I've got 5 pcs. of x on my board and the price for 1-20 ist lets say 0.05USD so thats 6 x 0.05 = 0.3USD! JLC has a minimum quantity for some if not the [...]

Right, I get that. 
What i meant was that the minimum quantity actually *is* in the table; it's the very first cell, namely the number '20'. Notice that the first cell sais "20-180: 0.004724409". The multiplier here is '20'. If you order 21 pieces, they charge you with 40. 

Regardless, it only serves as a cost *indication*. The whole point is that you can focus on the expensive components. The moment you place a JLC order, the prices and stock have changed anyways; they fluctuate too fast.

> I could imagine a "cost" button which shows a popup with the prices like this in a table: [...]

While I appreciate the thought, I don't think that's going to work in practice. If you have a few hundred components on your board, it just means clicking through a hundred popups and remembering all prices. I would like to sort.

Anyways, it could be that my feature request here is not useful at all. Cost is aggregated at the JLC part number, not at the designator - while the plugin is focused at the latter. I might just have to live with the fact that it's not here.

hattesen, 2023-01-16:
You might get some mileage from using the [KiCost](https://github.com/hildogjr/KiCost) (plugin and CLI utility) for obtaining component costs (documentation [here](https://hildogjr.github.io/KiCost/docs/_build/singlehtml/index.html)).
While it doesn't directly support JLCPCB component database and costs it supports LCSC (which have identical components and similar costs), as well as a large range of international component suppliers.
KiCost provides quantity-adjusted component costs, aggregated from all selected suppliers, and generates a spreadsheet, allowing you to compare costs of components across the selected suppliers.