Bouni/kicad-jlcpcb-tools #125: Drill Map File not generated with Generate fabrication files

MasterSpoon, 2022-02-03:
**Describe the bug**
Not sure if this was to be submitted as a bug or a feature request but at the moment with Generate Fabrication files the Drill Map Files are not created. This is listed as optional but suggested in JLCPCB's tutorial on how to generate files in KiCad. 

https://support.jlcpcb.com/article/149-how-to-generate-gerber-and-drill-files-in-kicad

"This drill map file provides additional information for drill holes, it is for human reading, it indicates which holes are plated and which are not, it also indicates total slotted holes. More information, less probability of error. "

Less probability of error is always a good thing and I have come across a few content creators that have got boards in the past without holes and cutouts. Even if ignored by JLC then can be useful when checking the fabrications files before submitting orders.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate fabrication files from JLCPCB Tools
2. review output files and NPTH-drl_map.gbr and PTH-drl_map.gbr are not created

**Expected behavior**
Drill map files to be created and included in fabrication output to reduce probability of errors.

# KiCad Version

```
Application: KiCad (64-bit)
Version: (6.0.0), release build
Libraries:
    wxWidgets 3.1.5
    libcurl/7.78.0-DEV Schannel zlib/1.2.11
Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Date: Dec 24 2021 19:17:02
    wxWidgets: 3.1.5 (wchar_t,wx containers)
    Boost: 1.76.0
    OCC: 7.5.0
    Curl: 7.78.0-DEV
    ngspice: 35
    Compiler: Visual C++ 1929 without C++ ABI
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

Bouni, 2022-02-03:
I've activated the drill map file creation

Bouni, 2022-02-03:
Will be part of the next release! I just wait for #124 to be merged befor I'll release.

MasterSpoon, 2022-02-06:
PR 126 appeared to have half fixed the issue, they are now generated in the output gerber folder, but not included in the zip file.

Bouni, 2022-02-07:
You're absolutely right! I overlooked that. Fix incoming in the next hours 😉

closed 2022-02-07 by Bouni