Bouni/kicad-jlcpcb-tools #127: Shot part details not working

nickoe, 2022-02-05:
**Describe the bug**
Hen I click "Show part details" I get a waiting cursor. Starting Kicad in a shell I get this trace back:

```
Traceback (most recent call last):
  File "...3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 727, in get_part_details
    dialog = PartDetailsDialog(self, part)
  File "...3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/partdetails.py", line 114, in __init__
    self.get_part_data()
  File "...3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/partdetails.py", line 226, in get_part_data
    self.get_scaled_bitmap(
  File ...3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/partdetails.py", line 130, in get_scaled_bitmap
    image = image.Scale(width, height, wx.IMAGE_QUALITY_HIGH)
TypeError: Image.Scale(): argument 1 has unexpected type 'float'
```
**To Reproduce**
Steps to reproduce the behavior:
1. Click "Show part details" anywhere
2. Busy cursor, no dialog appears.

**Expected behavior**
Dialog appears with content.

# KiCad Version

```
Application: KiCad

Version: (6.0.1), release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.81.0 OpenSSL/1.1.1m zlib/1.2.11 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.2 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.10.0 nghttp2/1.46.0

Platform: Linux 5.15.10-arch1-1 x86_64, 64 bit, Little endian, wxGTK, i3, x11

Build Info:
	Date: Jan 15 2022 23:16:29
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.78.0
	OCC: 7.5.3
	Curl: 7.81.0
	ngspice: 35
	Compiler: GCC 11.1.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2022-02-05:
I know what's the problem, maybe I can fix it later this evening

Bouni, 2022-02-07:
Not exactly same day but hey 😏

closed 2022-02-07 by Bouni

nickoe, 2022-02-09:
@Bouni  Sorry for the slow response, but I can confirm that your fix appears to fix the issue!

I test it manually by cloning the repo, maybe a new release tag is needed`?

Bouni, 2022-02-09:
@nickoe A release is on its way but I want to wait for some more fixes to be confirmed to avoid to release to often