Bouni/kicad-jlcpcb-tools #128: KiCad freezes when pressing plugin icon

Martinnnnnnnnnnn, 2022-02-06:
**Describe the bug**
When I press the JLCPCB tools icon in the PCB editor, KiCad freezes.

**To Reproduce**
I tried it with various projects, same result overall. I also tried the minimal case:
(0. Install plugin via plugin manager)
1. create new project
2. Add one resistor 0402 1k, connect both terminals
3. In PCB editor, import changes, draw one wire connecting both resistor pads
4. press JLCPCB icon

**Expected behavior**
Shouln't crash... should open plugin GUI

# KiCad Version
```
Application: KiCad PCB Editor (64-bit)
Version: (6.0.1), release build
Libraries:
    wxWidgets 3.1.5
    libcurl/7.78.0-DEV Schannel zlib/1.2.11
Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Date: Jan 15 2022 13:36:27
    wxWidgets: 3.1.5 (wchar_t,wx containers)
    Boost: 1.76.0
    OCC: 7.5.0
    Curl: 7.78.0-DEV
    ngspice: 35
    Compiler: Visual C++ 1929 without C++ ABI
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

Bouni, 2022-02-07:
Please repeat what you described above, but before you press the JLCPCB Icon, open the python console.
That will spawn a stderr/stdout windows which gives a stacktrace that shows what error causes your issue.

wowtah, 2022-02-07:
I have the same issue. Just installed the plugin, I clicked the icon, Kicad crashes.

yosva, 2022-02-07:
Hello. Same problem for me, nothing relevant I think in stderr, see attachment.
SO Windows 11, Kicad 6.0.1

[image]
[image]
e

Bouni, 2022-02-07:
I just did the exact same thing, deleted my plugin, installed it via PCM store.

Created a new project added two 0402 resistors with value 10k.
[image]

Created an outline connected te pads on the board:
[image]

Clicked the JLCPCB Icon and it worked without any error at all 😕 
[image]

And I run the exact same version:

```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.1), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan 15 2022 13:36:27
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-02-07:
Maybe some kind of permissen problem 🤔

Bouni, 2022-02-07:
Can one of you install an older version to see if its a problem with this specific version?

[image]

wowtah, 2022-02-07:
Uninstalled 2022.02.03 and installed 2022.01.06. Looks like it works now. Kicad no longer hangs, GUI window opens.

Martinnnnnnnnnnn, 2022-02-07:
> Please repeat what you described above, but before you press the JLCPCB Icon, open the python console. That will spawn a stderr/stdout windows which gives a stacktrace that shows what error causes your issue.

I just did that - no problem, GUI opens up. Tried also without opening the console, same result. Different project - same result, works!

Bouni, 2022-02-08:
@Martinnnnnnnnnnn you are still on 2022.02.03 and its working for you now, right?
@wowtah would you mind testing 2022.02.03 again and see if it works for you as well?

yosva, 2022-02-08:
2022.02.01 works for me, but 2022.02.03 and 2022.02.02 hangs Kicad

yosva, 2022-02-08:
[image]

Bouni, 2022-02-08:
I tested if it has to do with missing parts db and/or project db but I wasn't able to reproduce the error 😕

oofus, 2022-02-09:
I've been having the same problem in 6.0.1 on Windows 11. I've had to go back to 2022.01.06 for it to work. It seems to be newer versions than that freeze KiCad. Or at least do for me.

Bouni, 2022-02-09:
@oofus Do you get no error at all even if you start the console before clicking the JLCPCB icon?

Bouni, 2022-02-09:
I've just pushed a small change to main. I would highly appreciate if one or more of you guys could check if that solves the issue!

oofus, 2022-02-09:
@Bouni I get no error when the console is closed.
When the console is open the stdout/stderr window appears and I see similar output to what has already been posted here by others, ie the last output is about components already existing and will be ignored.

oofus, 2022-02-09:
@Bouni I've just downloaded latest from Github, having removed the version from the Plugin and Content Manager.
You're fix seems successful so far. I'm going to open and close everything a few times to see if it's consistent, as you think it may be a race condition.

Can I suggest that you put the plugin version in the plugin window (an about button?). I'm 99% sure I'm running what I downloaded from Github but it would be good to know 100%.

Bouni, 2022-02-09:
@oofus I think for quite some time how to put the version into the header. See #15 
I did not yet come up with a solution 😕

Martinnnnnnnnnnn, 2022-02-09:
> I've just pushed a small change to main. I would highly appreciate if one or more of you guys could check if that solves the issue!
Tried the plugin today (install via plugin manager): crash. Tried again, crash again.
Replaced mainwindow.py in some obscure user directory: crash again.
Opened console: crash again, no error shown in console. Just two messages DEBUG - update from board - Part R8 does not exist... and another for Q4. Nothing else.
Tried again with my test project (one resistor) Works. Debug message 2022.02.09 19:26:38 - DEBUG - download -  Download JLCPCB SMT Parts Library(20220208).csv with a size of 128.86MB
Going back to project that crashed before: Works! Messages
2022.02.09 19:28:19 - DEBUG - update_from_board -  Part Q4 does not exist in the database and will be created from the board.
2022.02.09 19:28:19 - DEBUG - update_from_board -  Part Q4 does not exist in the database and will be created from the board.
2022.02.09 19:28:19 - DEBUG - update_from_board -  Part Q3 does not exist in the database and will be created from the board.
2022.02.09 19:28:19 - DEBUG - update_from_board -  Part Q3 does not exist in the database and will be created from the board.
and so forth. Strange, R8 not shown again, every part twice.
If I had to guess my feeling would be it is something with loading the JLC database. Also the pattern is: After starting up Windows, it often fails. After it worked once, it works always.

Bouni, 2022-02-09:
> Tried the plugin today (install via plugin manager)

You need to install it manually as the PCM only shows released versions, not the working branch (main)

Martinnnnnnnnnnn, 2022-02-09:
> > Tried the plugin today (install via plugin manager)
> 
> You need to install it manually as the PCM only shows released versions, not the working branch (main)

Yes, I realized that. I copied in the modified mainwindow.py manually.

Bouni, 2022-02-09:
does it change the behavior?

Martinnnnnnnnnnn, 2022-02-09:
As I wrote above: no, crashes either way: Original installed via plugin manager or with modified mainwindow.py.
Works a few tries later, reason unknown.

closed 2022-04-01 by Bouni