Bouni/kicad-jlcpcb-tools #131: Edit/Copy LCSC number directly

Martinnnnnnnnnnn, 2022-02-07:
After selecting a part I realize: Oh, there's another identical component I missed. (Yes, I should have pressed "select alike" - maybe change that to a checkbox "always select alike before selecting part"?). It would be nice if one could copy the LCSC information directly, like copy/paste in the LCSC field in the table or double-clicking the field and entering the value manually.

Bouni, 2022-02-08:
Maybe I should make select alike a checkbox!?
I'll see what I can do about your direct enter / copy&paste feature request

ahalekelly, 2022-02-22:
Yes to all of these suggestions.
Another scenario it would be useful would be if you have the LCSC number already and want to just paste it in, like if you looked it up on https://yaqwsx.github.io/jlcparts/

Bouni, 2022-04-01:
This feature was implemented by @computergeek1507 in PR #140 🥳 

[image]

closed 2022-04-01 by Bouni