Bouni/kicad-jlcpcb-tools #132: Back-annotate LCSC number to schematic

Martinnnnnnnnnnn, 2022-02-07:
Normally I assign LCSC numbers already in the schematic, adding a "LCSC" field there. It would be nice if it would be possible to back-annotate the selected numbers from the plugin in the PCB editor into the schematic.

Bouni, 2022-02-08:
This was my goal from the beginning but is not possible at the moment!
To my knowledge there is no way to interact with the schematic in 6.0 but this is announced for 6.99 / 7.0.

uPesy, 2022-02-17:
Hey !
I have discovered your plugin recently. I have created a script that generate lcsc/jlc symbol and footprint library only with the part id. So if you used my script you can easily get the lcsc number in the schematic.

Bouni, 2022-02-18:
@uPesy just looked at your tool and, wow, I'm impressed 😁
If I understand it right your tool generates a footprint and a symbol from a given LCSC number which are added to a library and can be used in KiCAD then, right?
Is the LCSC number added to the symbol as an attribute?
What if I need let's say a 10k and a 4.7k resistor? Does the tool add two resistor symbols and footprints?

uPesy, 2022-02-25:
> @uPesy just looked at your tool and, wow, I'm impressed 😁 If I understand it right your tool generates a footprint and a symbol from a given LCSC number which are added to a library and can be used [...]

@Bouni 
Yes it generates symbol and footprint from a given LCSC id. I've just discovered the [TousstNicolas](https://github.com/TousstNicolas) repo that does the same thing.
The meta-datas are stored in symbol fields (in .lib files). For example, for the RP2040 chip :
```
DEF RP2040 U? 0 40 Y Y 1 L N
F0 "U?" 0 2300 60 H V C CNN
F1 "RP2040" 0 -2100 60 H V C CNN
F2 "LQFN-56_L7.0-W7.0-P0.4-EP" 0 -2200 60 H I C CNN
F4 "Raspberry Pi(树莓派)" 0 0 0 H I C CNN "Manufacturer"
F6 "C2040" 0 0 0 H I C CNN "LCSC Part"
F7 "Extended Part" 0 0 0 H I C CNN "JLC Part"
```
Is it possible from your plugin to grap the LCSC number from this fields ?

uPesy, 2022-02-25:
> What if I need let's say a 10k and a 4.7k resistor? Does the tool add two resistor symbols and footprints?

It will add 2 symbols in the lib file since LCSC number is different. Footprints are not directly linked to a symbol, it's symbols that are linked to footprints. So with your example, it will be only 1 footprint for the 2 resistors.
By the way, the script is not really needed for basic parts that are already in KiCad. It's better to add directly the LCSC id using your plugin GUI.

Bouni, 2022-02-25:
I think about a smart way to link symbol/footprint to LCSC number in case one matches many numbers 🤔

Bouni, 2022-07-26:
Implemented by @computergeek1507 in #198

closed 2022-07-26 by Bouni