Bouni/kicad-jlcpcb-tools #134: Text disappearing on high/DPI resolution screens

atlaste, 2022-02-08:
Yea, so in short, if I use my laptop I can't see half the text. It's probably because it's a 15" 4k screen. In display settings in Windows, I set the text size on 200%, otherwise my eyes would hurt too much :-)

Example:
[image]

Bouni, 2022-02-08:
Please post you PCB Editor version info. I implemented High DPI functionality recently and it works on Windows but not on linux due to a old wxWidgets version issue.

atlaste, 2022-02-09:
Now that I knew what to look for, I just installed the latest version of everything - and now it works fine. I'm just closing this.