Bouni/kicad-jlcpcb-tools #136: Issue with OpenSSL and KiCad 6.0 on Windows 11

krzysztofgawrys, 2022-02-08:
Plugin installed using Plugin and Content Manager and later manually, the same outcome.

After clicking on JLCPCB icon in PCB Editor popup window shows up:

OpenSSL: FATAL
OPENSSL_Uplink(...): no OPENSSL_Applink

Any solutions?

Bouni, 2022-02-09:
This seems to be ralated to the requests call to download the database 🤔 
Have you tried again since then?
Unfortunately I have no Windows 11 to test on ...

krzysztofgawrys, 2022-02-09:
Yes i did tried again, also i found out that other plugins have the same issue, when they try to download something.
So i think this is issue in KiCad itself, not strictly limited to your plugin, therefore i will close this issue and create new one on KiCaD github.

Cheers

closed 2022-02-09 by krzysztofgawrys

Bouni, 2022-02-09:
Thanks for reporting back!