Bouni/kicad-jlcpcb-tools #139: Question about sorting and grouping

Bouni, 2022-02-14:
Originally asked by @PMHK:

I have just discovered this plug-in and it looks like it will be incredibly useful. Having used it for a short time with KiCAD 6.0 on MacOS 11.6.2 I have a couple of questions/requests.

Currently if I click on any of the buttons to toggle BOM/POS the list of components refreshes back to the top of the list, this is quite frustrating when I am working through components at the bottom of a long list (the PCB I have been using to test this plugin has approx. 100 components) as I have to scroll back down every time I click the buttons?

On a related note, would it be possible to group the components according to type/footprint/value in a similar way to how KiCAD does in the Bulk Edit Fields option in the Schematic editor, this would make the length of the list more manageable reducing the impact of the above issue?

Apologies if there is a way to do this and I've missed it. If any further information is required then please let me know?

PMHK, 2022-02-14:
Bouni,

Many thanks for your response, I did puzzle a little as to whether to post as a discussion or an issue, but notice I shouldn't have to do that in future!

Sounds like you're already aware of the issue, if you do manage to solve it great, if not it's possible to live with it.

Kind regards,

Bouni, 2024-10-11:
Sorting is working since a few releases, grouping is not worth the effort in my eyes.

closed 2024-10-11 by Bouni