Bouni/kicad-jlcpcb-tools #141: Improve data table to fix multiple issues

Bouni, 2022-02-14:
Several issues did come up in the past days and months of which a lot are caused by the use of the `wxDataListCtrl`.

I looked around and found this demo `https://github.com/wxWidgets/Phoenix/blob/master/demo/DVC_DataViewModel.py` which I think is the way to go!

- Groupable columns for reference and value 
- Sortable
- Use of dataobjects which hopefully eleiminate the need to populate the list completely on every change.

I still need to look into it a bit more to make sure it solves all or at least most of the issues.

Bouni, 2022-02-14:
This has the potential to solve #139

wraybowling, 2022-07-15:
I'm interested to know if instead of bringing the functions of the spreadsheet to data list if we can bring the features of jlc tools to the spreadsheet. I don't know what direction kicad devs are headed for v7 though just a thought

Bouni, 2024-10-11:
Done :-)