Bouni/kicad-jlcpcb-tools #143: Wrong rotation and position of parts in JLCPCB in SMT assembly preview.

tibor-vavra, 2022-03-07:
Hi,
thank you for this useful plugin.

I face an issue with right placing and rotation of parts in jlcpcb Review Parts Placement
[image]

This is KiCad 3D Viewer
[image]

The board design looks OK. Is there any way how can I correct it or I'm doing something wrong?
Thank you for your help.

My KiCad version is:

Application: KiCad PCB Editor (64-bit)

Version: (6.0.2), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb 12 2022 14:32:09
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.6.0
	Curl: 7.78.0-DEV
	ngspice: 36
	Compiler: Visual C++ 1928 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

Bouni, 2022-03-09:
Thats a known problem with THT components. 
The KiCAD footprints often have there center of rotation not in their actual center but JLC's parts do.
At the moment, you can upload it as it is and JLC's engeneers will fix the placements for you at no cost.
I've no good idea how to overcome that issue at the moment :confused:

saschaludwig, 2022-03-21:
I can give you access to my KiCad files on my gitlab for further investigation.

Bouni, 2022-03-22:
@saschaludwig That would be really helpful!

Bouni, 2022-03-22:
I just tried to create a little dummy project but relized that the fuse shown in the screenshot seems not to be part of the KiCAD standard library.
Did you create the footprint by yourself?

Bouni, 2022-03-22:
Just to explain whats going on under the hood:

If you use a 1x4 pin header (KiCAD footprint PinHeader_1x04_P2.54mm_Vertical) you get this in KiCAD:

[image]

If you open the footprint in the editor you see that the center of the footprint is exactly on Pin 1:

[image]

If you assign LCSC number C390680 for example, generate the fabrication files and upload them to JLC you will see this:

[image]

So the rotation is off by 90° but also the center because JLC seems to set the center of rotation always to the center of the part and not the center of Pin 1!

I guess thats also the case for some SMD parts. @saschaludwig let me know which footprint you use and I'll take a look into it.

saschaludwig, 2022-03-22:
No, all parts are KiCad 6 standard parts.
And the rotation isn't the problem it looks like the created gerber + pos files created by the tools, trigger something that leads to a scaling somewhere along the JLCPCB processing. I will email you details about the project I have issues with.

Bouni, 2022-03-22:
I think I've seen that kind of issue before an it was something about imperial / metric settings 🤔 
Let me check, maybe I find the cause of this problem.

Also please post the version info of your KiCAD.

saschaludwig, 2022-03-22:
Okay, I found out that it has something to do with the origin of the files.
If I move the origin of the grid/placement to upper left corner of the board, I get distorted results like shown in the recent screenshots, even with manual export.
If I choose the placement origin to be in the lower left corner and have the coordinate system increasing on up/right I get matching results of footprints/layout.
So it has something to do with the origin of the files.

bvernoux, 2022-03-23:
I confirm JLCPCB preview is totally broken for connectors... I have asked them several time to fix it

Bouni, 2022-04-05:
I think there's no good solution for this at the moment. I will close this issue for now, you can re-open at any point in the future if you come up with a solution or if JLC offers one.

closed 2022-04-05 by Bouni