Bouni/kicad-jlcpcb-tools #144: When panalizing of duplicating the POS file does not generate POS X,Y for all components

DomoticX, 2022-03-08:
**KiCAD version**
6.0.2

**Operating system**
Windows 10

**Description**
When exporting the JLCPCB files, the POS file does not generate POS X,Y for all components on the screen/project (e.g.) when the board is copied/duplicated

**Steps to Reproduce**
CTRL+C the board
CTRL+V paste the board
remove edge cuts on the 2 boards
draw new edge cuts around the new boards
export with JLCPCB
POS file only has XY for components of 1 board (instead of two)

Bouni, 2022-03-09:
The problem here is most likely that you duplicate the parts so they have the same Reference.
If there is just a single Resistor on the board, lets call it R1 you end up with two R1 after duplicating the board.
My BOM / POS generation script was not written with that in mind.
I'll see if its easily possible to achive that!

Bouni, 2022-03-09:
@DomoticX could you test my PR and check if it fixes your problem?

DomoticX, 2022-03-09:
> @DomoticX could you test my PR and check if it fixes your problem?

Thank you for your support, i will check it out!

DomoticX, 2022-03-09:
> The problem here is most likely that you duplicate the parts so they have the same Reference. If there is just a single Resistor on the board, lets call it R1 you end up with two R1 after [...]

I know ;-). In Kicad 5...when exporting POS fabrication output, all components on the screen/board where outputted to the POS file correctly even if de REF of the components is the same...., the thing we need this feature is like when we panelize PCB's it needs to output all components.

DomoticX, 2022-03-09:
YES YES YES !!, thank you very much!, it is working perfectly now!, also CONFIRMED and tested @ JLCPCB, superb!

DomoticX, 2022-03-09:
I just bought you a nice coffee @ ko-fi.com ENJOY!

closed 2022-03-09 by DomoticX

Bouni, 2022-03-10:
Thanks for your support, highly appreciate it!

PR #146 (2022-03-10): Fix panalized pos files
This should fix #144